technyon/nuki_hub #180: Lock status not updated if manually operated

mmaccaroni, 2023-04-24:
Hi!
First of all thanks for this awesome component!
I noticed that if I manually operated the lock (either with the knob or the keys) or use the nuki app, I do not get updated status: it remains in the previously set one.
I use the latest version of the component, set as app over a nuki non-pro which is also connected to a nuki bridge.
If i reset the esp 32 of course the status is correctly reflected.
It is like every change in status not directly triggered by mqtt is not reflected.
Is it an expected behaviour? Everything else works smootly.

mundschenk-at, 2023-04-24:
@technyon Not sure if you have already seen it, but the Nuki devs just answered @I-Connect's question regarding the "register as app" feature: https://developer.nuki.io/t/difference-between-authorizing-with-id-type-app-or-bridge/20866/2

technyon, 2023-04-24:
That's the short version yes :)

closed 2023-04-24 by mmaccaroni