technyon/nuki_hub #183: Feature request

Eguar-rc, 2023-04-29:
Lock permissions, users, keypad, fob.
Controlling users or devices remotely would be useful.

technyon, 2023-04-30:
What exactly is your use case. There a lot of things covered by the ble API, adding all of those in an MQTT API could prove difficult. Do you just want to enable/disable existing authorizations?

Eguar-rc, 2023-05-01:
Thanks for your answer.

 

I am interested in being able to cancel access to a specific FOB, for example if I am outside my network and want to prevent the entry of a guest. 

Effectively I want to enable/disable existing authorizations.

Enviado el: domingo, 30 de abril de 2023 21:51
Para: technyon/nuki_hub ***@***.***>
CC: Eguar ***@***.***>; Author ***@***.***>

technyon, 2023-05-06:
Hi,

Just enabling or disabling existing authorizations should be possible, but it's not a small feature request. I'll look into and and see what I can do.

Eguar-rc, 2023-05-06:
Thanks, with that nuki_hub would cover everything that I previously did through the web.
Thank you also for sharing your work, I have applied it and it is working perfectly.

closed 2023-06-08 by technyon