technyon/nuki_hub #184: GPIO legend

Asino123, 2023-04-30:
Good morning, thank you for your wonderful work. I previously configured a nuki_hub with version 6. I did several tests and one of the options I used and which I found essential was the opening and closing via GPIO. The functionality had been tested successfully. Now looking at the guide I can't find where the GPIOs are indicated for opening and closing. Could you help me? Thanks.

technyon, 2023-04-30:
In the latest version, GPIO is configured by the user, you can select which PIN does what. Open the configuration portal and click Edit under "GPIO Configuration". Configure the PINs however you need them and click save. If you need a reference to the previous configuration, you can check the history of the README.md file on github.

Asino123, 2023-04-30:
Thanks a lot for the answer.  i will update my version.

technyon, 2023-05-06:
@Asino123 Did this resolve your issue?

Asino123, 2023-05-06:
hi, yes thank you very much.

technyon, 2023-05-06:
ok, closing issue

closed 2023-05-06 by technyon