technyon/nuki_hub #185: Supported ESP32 Version

tpftpf, 2023-04-30:
When I use the web installer I get the message:
"Your ESP32-C3 board is not supported."
Is there a workaround for this?
Thanks!

technyon, 2023-04-30:
Hi,

The C3 is not supported and there's no easy way to change this. The C3 is a single core ESP, and even when compiling for the, the firmware crashes. Considering how cheap ESPs are, it's not worth the effort.

tpftpf, 2023-04-30:
> From "Jan-Ole Schümann" ***@***.***>
> To "technyon/nuki_hub" ***@***.***>
> Cc "tpftpf" ***@***.***>; "Author" ***@***.***>
> Date 30.04.23 10:34:23
> Subject Re: [technyon/nuki_hub] Supported ESP32 Version (Issue #185)

Hi,

maybe changing "The NUKI Hub software runs on any ESP32 module" could 
clarify things.

Thanks!

technyon, 2023-04-30:
Yes it's misleading. I have to think how to better specify the compatible ESP modules, since there a a lot of variants under a lot of names out there.

achint-s, 2023-05-05:
While we are at it. I have experienced instability on a few of the ESP32 dev kits that I have used. Will something like an m5stack Atom Lite work in wifi only mode? Seems much better built.

technyon, 2023-05-06:
Yes they're a good choice since the integrated antenna is better than the PCB antennas that many other ESPs use.

closed 2023-05-06 by technyon