technyon/nuki_hub #186: Feature Request - HTTP REST Interface

patrickmarchhart, 2023-05-01:
Is there a possibility to control the Nuki Hub via HTTP commands?

patrickmarchhart, 2023-05-01:
I would like to integrate the Nuki Hub into a Loxone automation system.

technyon, 2023-05-04:
To give you an answer, such feature isn't impossible, but it's a considerable amount of work. It's not only implementation, but also designing an API and testing if the API works as intended. I've chosen MQTT as a protocol because it's quite easy to use and something that's supported by most IOT solutions, REST certainly needs more experience to use it.

As far as I've seen, there are ways to integrate MQTT into Loxone, which is I think easier and faster than integrating a REST API. So no, I'm not planning to add a REST API at this time.

closed 2023-05-15 by patrickmarchhart