technyon/nuki_hub #188: OTA functionality not ready. Please wait a moment and reload. (from 8.22 to 8.23)

fisch55, 2023-05-07:
Hi, any Idea how can I update from 8.22 because getting this message....

OTA functionality not ready. Please wait a moment and reload.

technyon, 2023-05-07:
Post the content of the system information page please.

fisch55, 2023-05-07:
`NUKI Hub version: 8.22
run: true
deviceId: xxxxx
mqttbroker: xxxxxx
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: true
resttmr: -1
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
regAsApp: false
nrRetry: 5
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: 19
macb1: -121
macb2: -121
MQTT connected: Yes
Lock firmware version: 2.14.5
Lock hardware version: 11.1
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: Yes
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 0 minutes
Heap: 90596
Stack watermarks: nw: 6104, nuki: 664, pd: 236
Restart reason FW: ConfigurationUpdated
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.`

technyon, 2023-05-10:
Hi,

Sorry for the late answer I was sick unfortunately.

If you get this message, it means for whatever reason your ESP is a reboot-loop. The only option that could trigger this in your case is "Restart on disconnect". Try to disable it and try again.

technyon, 2023-05-13:
@fisch55 Did this solve your issue?

fisch55, 2023-05-13:
Sorry for late answer. It works ….. can be closed 👍

closed 2023-05-14 by technyon