technyon/nuki_hub #190: Olimex boards generating command timeouts

ruimarinho, 2023-05-21:
Hi @technyon,

Upgraded to 8.23 on my Olimex board coming from a very old version on a WiFi ESP32 and on the exact same place (next to the door), I'm getting timeouts now:

```
2023-05-21T11:50:00+0000 : 0 : 1 : nuki/maintenance/uptime 0
2023-05-21T11:50:07+0000 : 0 : 0 : nuki/lock/action lock
2023-05-21T11:50:07+0000 : 0 : 0 : nuki/maintenance/log Lock action received: lock
2023-05-21T11:50:07+0000 : 0 : 1 : nuki/lock/action ack
2023-05-21T11:50:20+0000 : 0 : 1 : nuki/lock/commandResult timeOut
2023-05-21T11:50:20+0000 : 0 : 0 : nuki/maintenance/log Lock action result: timeOut
2023-05-21T11:50:20+0000 : 0 : 0 : nuki/maintenance/log Lock: Last command failed, retrying after 100 milliseconds. Retry 1 of 3
2023-05-21T11:50:20+0000 : 0 : 1 : nuki/lock/retry 1
2023-05-21T11:50:20+0000 : 0 : 1 : nuki/lock/commandResult success
2023-05-21T11:50:20+0000 : 0 : 0 : nuki/maintenance/log Lock action result: success
2023-05-21T11:50:20+0000 : 0 : 1 : nuki/lock/retry --
2023-05-21T11:50:21+0000 : 0 : 1 : nuki/lock/query/lockstateCommandResult success
2023-05-21T11:50:21+0000 : 0 : 1 : nuki/lock/state locked
2023-05-21T11:50:21+0000 : 0 : 1 : nuki/lock/binaryState locked
2023-05-21T11:50:21+0000 : 0 : 1 : nuki/lock/lastLockAction Lock
2023-05-21T11:50:21+0000 : 0 : 0 : nuki/maintenance/log Querying lock state: locked
2023-05-21T11:50:30+0000 : 0 : 1 : nuki/maintenance/uptime 1
```

It's happening for pretty much all commands. I'm running Nuki firmware 3.6.9.

technyon, 2023-05-21:
There are some reports like this especially when unpairing and pairing locks again. What often helps is to erase the ESP and re-flash NUKI Hub and then to try again.

ruimarinho, 2023-05-21:
Should I unpair first on the esp before erasing + reflashing? It took me several attempts to get pairing to work with this board..

technyon, 2023-05-22:
Hi. No need to unpair. If you erase the ESP, all pairing information is deleted anyway, so you'll have to re-pair. I know it can take a bit to pair, not sure why sometimes it's really and at other time it seems it just doesns't want to pair.

ruimarinho, 2023-05-22:
So the Nuki Lock itself does not store any pairing information? Interesting.. I've ran into that situation where it's either immediate in pairing or very difficult to do so.

Meanwhile looking at the logs here's what I see (let me know if you want me to open another issue):

```
2023-05-22T20:26:49+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:26:49+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:26:52+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:26:52+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:26:54+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:26:54+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:01+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:01+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:01+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:01+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:01+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:01+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:04+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:04+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:07+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:07+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:10+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:10+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:12+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:12+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:14+0000 : 0 : 1 : nuki2/maintenance/uptime 1935
2023-05-22T20:27:15+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:15+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:17+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:17+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:20+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:20+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:23+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:23+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:26+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:26+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:29+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:29+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:32+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:32+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:34+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:34+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:38+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:38+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:40+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:40+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:43+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:43+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:44+0000 : 0 : 1 : nuki2/maintenance/uptime 1935
2023-05-22T20:27:46+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:46+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
2023-05-22T20:27:48+0000 : 0 : 1 : nuki2/lock/query/lockstateCommandResult success
2023-05-22T20:27:48+0000 : 0 : 0 : nuki2/maintenance/log Querying lock state: unlocked
```

technyon, 2023-05-22:
Did you enable "Register as app"? That's not recommended and unless you also run a bridge it's result in endless polling of the lock state.

ruimarinho, 2023-05-22:
Yes as this is my second unit. I’ll do the erase + register as a bridge normally tomorrow then. Thanks for the insight!

ruimarinho, 2023-05-23:
@technyon that seems to have made the trick. Erased and re-configured everything and now I don't see any timeouts happening. Pairing was still very difficult but this time around I decided to be a lot more patient and just let it stay on "Nuki lock start pairing" for much longer time.

Thanks for your support again!

iranl, 2024-02-11:
@technyon: Seems to have been resolved. Close?

technyon, 2024-02-11:
Yes can be closed

closed 2024-02-11 by technyon