technyon/nuki_hub #191: Emulate door sensor

mbrevda, 2023-05-28:
The Nuki door sensor allows the unlatch duration to [start only after the door has been opened](https://support.nuki.io/hc/en-us/articles/4406386190225-Adapted-features-of-the-Smart-Lock-for-more-efficient-control#:~:text=Once%20the%20Door%20Sensor%20has%20registered%20that%20the%20door%20has%20been%20opened%2C%20the%20latch%20is%20released%20and%20you%20manually%20can%20close%20the%20door%20again.). Can this project be used to send the door state to the Smart Lock and effective emulate the door sensor?

My use case is that I have my own door sensor, and I'd rather not have to purchase Nuki's.

I've tried publishing `doorOpened` to `lock/doorSensorState` but that didn't seem to work (the door state in the Nuki app didn't update).

Thanks!

technyon, 2023-05-28:
Hi,

No this not possible, the bluetooth api has no such functionality. You can replace your current door sensor with the NUKI door sensor and read the sensor state via NUKI Hub ... so it's read-only, writing back the door sensor state is not possible.

mbrevda, 2023-05-28:
shame - I thought the Nuki sensor worked via bluetooth and there would be some way to emulate that.

Thanks!

closed 2023-05-28 by mbrevda

technyon, 2023-05-28:
It does work via bluetooth, however it's paired to the lock, the lock reports its state to NUKI Hub. Since the lock has the ownership of the door sensor, there's nothing NUKI Hub can do unfortunately.

mbrevda, 2023-05-28:
> however it's paired to the lock

Right - so it would be great it we can emulate that connection via an esp32 and push out desired lock state

technyon, 2023-05-28:
To do that you'd need to

1. Emulate a door sensor so the ESP32 can appear as a door sensor to the lock
2. Emulate a lock so that the ESP32 appears to the door sensor as a lock

Unfortunetely it's undocumented how the sensor and the lock talk to each other. On top of that, while emulating a lock isn't impossible, it would a a substantial effort to do so.

mbrevda, 2023-05-28:
> Emulate a lock so that the ESP32 appears to the door sensor as a lock

Couldn't the emulated door sensor connect to the real lock?

> Unfortunetely it's undocumented how the sensor and the lock talk to each other

Sadlt. And I think their sensor is credulously priced (in addition to not being available for purchase around here)

technyon, 2023-05-28:
> Couldn't the emulated door sensor connect to the real lock?

Yes, however as I said they didn't publish any documentation on that part. I'd need to know how the lock and the door sensor talk to each other. Plus you pair the via the app, I'd need to know how that works too.

mbrevda, 2023-05-28:
Got it. Thanks!

iranl, 2025-08-01:
No.

It is still on the backlog, but has very low priority.

In part because it is undocumented and in part because there is only direct communication between the door sensor and lock and we don't have access to their code. 

The documented parts of the API give some insight into how the lock and door sensor might communicate (encryption/pairing works probably the same as lock-hub/app) but it will still be difficult to emulate a sensor/keypad/fob.

technyon, 2026-08-15:
Thanks will have a look. Still it won't be all that easy I guess - the ESP would need to be paired, not as a bridge I think

mbrevda, 2026-08-16:
Surely the bridge is paired already?

technyon, 2026-08-17:
Yes it's paired, but not as a door sensor. Not sure if it would work if paired as bridge.

mbrevda, 2026-08-17:
In the esphome project it's also not paired as a door sensor... but only tested paired as an app

technyon, 2026-08-17:
ok thanks

antoniodgonzalez, 2026-09-02:
I'm trying this feature, but not sure how it should work. I would expect that I need to pair the emulated door sensor to the lock first, but I don't know if it is possible to put the esp in paring mode once it is already paired to the lock.

In mqtt I see doorSensorState = unavailable
Expected as it is not paired?

I can write 0 / 1 to doorSensorOverride and get success; but doorSensorState remains unavailable. error would be expected in this case?

technyon, 2026-09-03:
Hi, no need to pair as door sensor. Even paired as a bridge, Nuki can send the command. You will not see a door sensor state in the app, as there's no door sensor paired.

antoniodgonzalez, 2026-09-03:
Hi, thanks @technyon 
I was expecting the lock to do not autolock while the door is open. But this is not the behavior I get after writing doorSensorOverride with success. Maybe something I'm missing?

mbrevda, 2026-09-03:
Nuki places a 30 second hard limit after which it always locks (failsafe). After how long were you experiencing a lock?

antoniodgonzalez, 2026-09-03:
It is locked after the time I have configured in Autolock / Lock after / 30 seconds. As I understand from [here](https://help.nuki.io/hc/en-001/articles/4407476465937-Auto-Lock-Automatically-lock-your-door), it should be locked only if the door is closed?
In advancedJson, I have autoLockEnabled and immediateAutoLockEnabled set to 1, but the lock is not auto-locked after setting doorsensorOverride, but after the timeout.

mbrevda, 2026-09-04:
Seems this is the intended behavior 

https://developer.nuki.io/t/nuki-3-0-pro-locks-door-with-door-open-sensor/19493

antoniodgonzalez, 2026-09-04:
Thanks @mbrevda, understood. I already had an automation to unset autoUnlock when the door is open to get the behavior I needed, just though I could simplify this with the door sensor emulation.

What is your usecase for the sensor emulation? Did you manage to make immediateAutoLockEnabled work?

mbrevda, 2026-09-04:
Yes, to an extent. Depending on the setup (if a bridge is involved, etc) "immediate" is relative, but the sensor allowes to lengthen the door unlatch time and usually will re-latch is pretty quickly after the door was openwd. As of now, about 4-7seconds later, ~80% of the time... 

From the forums it would seem that the original sensor isn't any/much faster (if someone has one we can learn from it and see how this can be improved).

technyon, 2026-09-04:
The door is locked even if the door sensor reports the door is open. Due to Nuki this behavior has been changed due to user request, as sometimes the lock misses the update from the door sensor so for some users the door stayed unlocked. Quite annoying I have to say.