technyon/nuki_hub #193: Unlach Help

CAVACO-PT, 2023-05-29:
Hi , thank you for the work on this , its realy better than the bridge ...

i have disconnected one of my locks from the bridge and pair with with a m5 atom POE , everyhing is working inside mqtt and HA , but i dont have any unlach options , its the one of the most used features i need , 

can someone point me out in the right direction to get this working , im been looking and dont see anything wrong 

[image]

CAVACO-PT, 2023-05-29:
ok , seams the command is not present , but the service works 👍 

[image]

mundschenk-at, 2023-05-29:
It is, that is the HA standard way of effecting that action. You can always add helper button that calls the service if you want that specific UI.

CAVACO-PT, 2023-05-29:
Anyone knows if this can handle mutiple locks ?? i had problems with the nuki bridge handling multiple locks , due to it being limited handling mutiple requests , can i add here my 2 locks without any issues ??

mundschenk-at, 2023-05-29:
You need a separate ESP per lock.

closed 2023-07-24 by technyon