technyon/nuki_hub #195: MQTT connection issues with SL 3 since v8.24

bcutter, 2023-06-09:
Since I updated to Nuki Hub 8.24, one of my Nuki Hub serving a SL 3 (the other one with a SL 3 Pro is completely fine) is quite unreliable looking at the `binary_sensor.lock_name_mqtt_connected` entity in HA:

[image]

Nothing else has changed. I even try to monitor if the ESP connects to different APs and if that could be an issue, up to now I could not discover this as the root cause. At the same time looking at https://github.com/technyon/nuki_hub/releases/tag/8.24 release notes I can not spot anything which could be directly related to this.

Update: looking at the custom uptime sensor and comparing it, it looks like a general connection issue (WiFi connection), doesn't it?

good one without MQTT connection issues:
[image]

problematic one with MQTT connection issues:
[image]

technyon, 2023-06-09:
Can you confirm this is related to 8.24 by going back to 8.23 and logging the connection state?

Updating the arduino core for esp32 could make wifi less reliable, although I don't think so, updates usually improved stability. Also there's nothing regarding wifi in the changelog:

https://github.com/espressif/arduino-esp32/releases/tag/2.0.9

bcutter, 2023-06-10:
I downgraded to 8.23 now (only the affected ESP) and will report back in a few days.

bcutter, 2023-06-17:
Short update:

SL 3 Pro with Nuki Hub 8.24:
[image]

[image]

SL 3 with Nuki Hub 8.23:
[image]
(no recent incidents, last one 3 days ago)

[image]

---

So looking at MQTT connection entites and Nuki Hub only I tend to say "things got worse from 8.23 to 8.24". Comparing the uptime sensors it looks like it's more a general (WiFi?) connection/stability thing.

alexdelprete, 2023-06-18:
I have no issues regarding MQTT connection, I'm using EMQX.

[image]
[image]

alexdelprete, 2023-06-18:
@mcutter what broker are you using? can you produce serial debug logs so we can see exactly what happens when NH does the MQTT connect?

technyon, 2023-06-18:
Also please post the system information page, best case of both ESPs so we can compare.

bcutter, 2023-06-18:
1. I use the default addon in Home Assistant as MQTT broker (Mosquitto).
2. Logs sounds like hard work as the logging possibilities are quite limited, for usable (long-term) logs I'd probably first do a deep-dive into it. The web only shows limited number of log entries.
3. System information pages
- SL 3 Pro with NH 8.24
```
System Information

NUKI Hub version: 8.24
run: true
deviceId: XXXXXXXXXX
deviceIdOp: XXXXXXXXXX
mqttbroker: xxx.xxx.xxx.xxx
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki_door_a
openerena: false
mqttoppath: 
maxkpad: 4
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: xxx.xxx.xxx.xxx
ipsub: 255.255.255.0
ipgtw: xxx.xxx.xxx.xxx
dnssrv: xxx.xxx.xxx.xxx
nwhw: 1
rssipb: -1
hostname: ESP-Nuki-Hub-Door-A
nettmout: -1
restdisc: true
rstbcn: 60
lockStInterval: 3600
configInterval: 3600
batInterval: 7200
kpInterval: 1800
kpEnabled: true
accLvl: 
regAsApp: false
nrRetry: 5
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: true
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.6.9
Lock hardware version: 5.11
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: Yes
Network device: Built-in Wifi
Uptime: 1456 minutes
Heap: 84840
Stack watermarks: nw: 6104, nuki: 312, pd: 248
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```
- SL 3 with NH 8.23
```
System Information

NUKI Hub version: 8.23
run: true
deviceId: YYYYYYYYYY
deviceIdOp: YYYYYYYYYY
mqttbroker: xxx.xxx.xxx.xxx
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki_door_b
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: xxx.xxx.xxx.xxx
ipsub: 255.255.255.0
ipgtw: xxx.xxx.xxx.xxx
dnssrv: xxx.xxx.xxx.xxx
nwhw: 1
rssipb: -1
hostname: ESP-Nuki-Hub-Door-B
nettmout: -1
restdisc: true
rstbcn: 60
lockStInterval: 3600
configInterval: 3600
batInterval: 7200
kpInterval: 1800
kpEnabled: false
accLvl: 
regAsApp: false
nrRetry: 5
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: true
pubdbg: false
prdtimeout: 60
hasmac: true
macb0: -XX
macb1: YY
macb2: ZZZ
MQTT connected: Yes
Lock firmware version: 3.6.9
Lock hardware version: 4.10
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 232 minutes
Heap: 90124
Stack watermarks: nw: 6104, nuki: 600, pd: 248
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

The last MQTT connection issue for door B has been many days ago meanwhile running 8.23.
I updated NH once again to 8.24 on that ESP now, let's see if things change.

mundschenk-at, 2023-06-18:
`screen -L /dev/<your-usb-tty> 115200` and then send us `screenlog.0` (ESP32 connected via USB to the computer of course, UNIXish OS).

technyon, 2023-06-20:
I agree that getting serial logs would help to get some insight into what's going on. They would especially tell if the MQTT connection is really interrupted.

technyon, 2023-06-28:
You could the binary from this issue, maybe it's related:

https://github.com/technyon/nuki_hub/issues/196

bcutter, 2023-07-08:
Updated to 8.25 now, will monitor it for a few days. With 8.24 it seems like the ESP itself had more MQTT/connection issues when connected to a certain WiFi AP while others were working better/fine. That behaviour is/was definitely new and did not exist in the past.

Not sure if that makes sense (can Arduino Core cause such WiFi and in the end MQTT connection issues?), just a pre-summary form my current point of view.

technyon, 2023-07-09:
Hi,

Yes the arduino core can cause such issues. It's the code from Espressif that everything else is built upon, including drivers for wifi, bluetooth, TCP/IP stack. If anything is broken in there, it can cause any issue. We did test it before release 8.24, but couldn't observe the issue, only now with longer testing we could see it.

closed 2023-07-24 by technyon