technyon/nuki_hub #198: Bluetooth connection lost

AndLindemann, 2023-06-21:
Looks like I need to reboot my M5Atom around every 3 hours or so because it loses the bluetooth connection to the lock.
It seems to work well with the "Restart if bluetooth beacons not received" set to 60 seconds so far though, so not a huge issue right now.
Just checking if this is a known issue or something that can potentially be fixed?

```
NUKI Hub version: 8.24
run: true
deviceId: 1115837375
deviceIdOp: 1115837375
mqttbroker: 10.0.70.60
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: true
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: false
rstbcn: 60
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.6.9
Lock hardware version: 4.10
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: Yes
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 185 minutes
Heap: 85232
Stack watermarks: nw: 6104, nuki: 616, pd: 244
Restart reason FW: BLEBeaconWatchdog
Restart reason ESP: ESP_RST_INT_WDT: Reset (software or hardware) due to interrupt watchdog

```

technyon, 2023-06-21:
Hi,

this indeed shouldn't happen but it's very hard to tell what's going on with having access to your ESP. The BLEBeaconWatchdog restart happens when the ESP doesn't receive any bluetooth beacons as a measure of wether or not the bluetooth stack is still alive. So either  the bluetooth stack has died, or the ESP didn't indeed receive beacons due to bad reception. How close is your ESP to the lock?

AndLindemann, 2023-06-21:
The ESP is about 3m from the lock with direct line of sight. I haven't seen any signs of connection issues, it usually works fine without any delays and the connection loss seems to always occur after at least 4 hours (currently I have an uptime of 8 hours actually without any issues), it is actually quite consistent in that way. Looks more like something that builds up over time until it eventually fails, if you know what I mean.
It doesn't really impact functionality much due to the automatic reboot so is more like a cosmetic issue right now, nevertheless of course it'd be nice if it would just run stable. :) 
I could try attaching it to a terminal to get logs, though the next PC is upstairs and based on the experience with my bluetooth headset, connectivity from there to the door is not great, so I'd have to see eventually uninstall the lock and take it upstairs as well - if you think that there's anything worthwhile in the logs.

AndLindemann, 2023-06-22:
Out of curiosity, I replaced the m5atom-lite with a different m5atom-lite as I still have some spare lying around. Factory reset the lock, installed everything again. Placed the ESP a bit closer to the lock this time (overall bluetooth RSSI has improved slightly but is generally already very good in the low 60ies).
It ran fine for about 5 hours before it lost bluetooth and the BLE watchdog rebooted it.
I don't think it's the actual bluetooth reception that's the issue here, probably more that the bluetooth stack is just dying after some time.
That seems odd, as I have a couple of m5atom-lite working as Espresense devices around the house and none of that ever had any bluetooth issues at all.

mundschenk-at, 2023-06-22:
Yeah, the Bluetooth stack apparently dying every once in a while is unfortunately normal behavior. I suspect it stems from using both WiFi and Bluetooth extensively. See https://github.com/technyon/nuki_hub/issues/28 for background on the development of the BLE watchdog.

AndLindemann, 2023-06-22:
Yeah that seems like what's happening.
Wondering if Espresense (https://github.com/ESPresense/ESPresense) uses a different BLE library, as the m5atom-lite's running Espresense (running 6 of them here) never seem to have an issue with parallel WiFi + BLE operations, but then the Espresense is of course more or less only listening on the BLE and not actually communicating with other bluetooth devices.

AndLindemann, 2023-06-22:
Any possibility to publish the "uptime" of the ESP to a home assistant sensor like you already do for the RSSI values? I'd like some record to see when and how often the ESP gets reset by the watchdogs and as it happens very fast usually, you can't really see it from the other published metrics.

AndLindemann, 2023-06-22:
Not sure if this should go on a separate issue, but I've also seen these now on 2 different m5atom-lites:

```
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
```

Of course that may or may not be related to Bluetooth, but at least it wasn't the watchdog rebooting the ESP this time. Not sure if this can be caught in the logs?

mundschenk-at, 2023-06-22:
Yes, that's when something unforeseen happens (different issue, though).

technyon, 2023-06-22:
I'd say don't worry too much about it, we made at least sure the ESP reboots whenever the bluetooth stack dies. If you want to fix it, you could switch to a wired ethernet solution which doesn't suffer from these issues.

Adding the uptime sensor should be possible.

mundschenk-at, 2023-06-22:
```yaml
mqtt:
  sensor:
    - name: "Nuki Hub Uptime"
      unique_id: '<some unique id>'
      state_topic: "<prefix>/maintenance/uptime"
      qos: 1
      device_class: duration
      unit_of_measurement: min 
      device:
        identifiers: nuki_<lockid>
      entity_category: diagnostic
```

AndLindemann, 2023-06-22:
Yeah I probably would, but the next available ethernet port is a few meters in the wrong direction and with a not-too-thick wall in the line of sight to the lock. I'd need to test the bluetooth connection from there first before eventually considering an ESP with ethernet.
Though I'd agree, as long as the ESP reboots within seconds when it encounters an issue, it probably is no big thing during day-to-day operations.

@mundschenk-at: Damn, I should've checked. I wasn't aware we already have uptime send to MQTT. Yeah, then I can just add a custom sensor to HA, no big deal. Good finding. :) 

I guess I'll close this for now, thanks for you input both. :)

closed 2023-06-22 by AndLindemann