technyon/nuki_hub #199: Installation on Shelly 1 Plus

Gobbo28, 2023-06-21:
Hello,

I was wondering if anybody has ever tried to install the Firmware on a Shelly 1 Plus?
I tried today. Flashing it via the Web installer was successful, but it seems that the ESP32 isn't able to boot the firmware.

technyon, 2023-06-23:
Shellys use a single core variant of the ESP32, so the binary won't run on those. You'd need to compile it specifically for the single core variant, which is somewhat complicated. I'm thinking about how to release binaries for several variants, but I'd some automation to do that. For example I've tried running it on the new C3, but that one needs a specific binary too (although it is a bit easier to compile this one).

Gobbo28, 2023-06-23:
Ah ok, I understand. Thanks for the explanation.

Would be a cool Feature to have the Firmware working on such a device, since I have a shelly in my entrance area.

lucaoliano, 2023-07-16:
Hello,
for Shelly 1 Plus you can use this firmware https://github.com/lucaoliano/nuki_hub/suites/14334003703/artifacts/806776942

Basically it's just the technyon firmware built for esp32solo1 without any additions.

KingSausage, 2023-10-14:
I really like the idea. How does this work? Do you flash the whole Shelly and the stock firmware will be replaced or does it work as an add-on? And is this OTA possible?
Sorry for the questions, I have no experience with the Shelly Flashing prodecure and I would like to keep the stock fw for "normal". Would it also be possible to get this working on a Shelly Plug?

technyon, 2023-10-15:
This is not an addon, you have to flash this firmware and it will replace the default shelly firmware. Not sure about the flash procedure. I think some can flash a 3rd party via the web interface, others you'll have to open the shelly and connect a programmer.

KingSausage, 2023-10-15:
I was wondering because Gobbo28 said, he was already running a shell in his entrance area. So I assumed he wanted to keep the function plus the nuki-hub. I checked the file and as it is a python script, you cannot flash it OTA. I think I need to connect the shelly like every ESP.
I wanted to try it with a shelly plug, but it didn't accept the zip file:)

lucaoliano, 2023-10-15:
If you unzip the file from the previous post, you can find a sh script with the flash command. I report the command here: `esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq keep --flash_size detect 0x1000 bootloader.bin 0x10000 nuki_hub.bin 0x8000 nuki_hub.partitions.bin`. Change the `--port` according with your environment. The firmware can be flashed via the Shelly debug port, you need an USB to TTL adapter. [Here](https://templates.blakadder.com/shelly_plus_1.html) you can find the description of the port pins. Before flashing the nuki_hub firmware onto the Shelly, I recommend to  backup the stock firmware, but I cannot guarantee that by restoring the stock firmware the Shelly will still be able to connect to the Shelly cloud.

The commands to backup and restore the stock firmware are:

- `esptool.py read_flash 0x0 0x400000 flash_4M.bin`
- `esptool.py write_flash 0x0 flash_4M.bin`

Remember to backup the firmware of each shelly you intend to modify because each one has a different key inside for cloud authentication.

KingSausage, 2023-10-15:
Thank you very much for this detailed tutorial!
I have already looked into the zip file and realized, that this is an own flash procedure for a esp32solo. What I am curious about is, when the thread starter has already a shelly Plus 1 built e.g. behind a switch and now flashes the nukihub, he looses the ability to control any switch/loads, am I right? So it is not more than an ESP behind the wall connected to 230V, am I right?

As I am still waiting for my ttl USB board,my first attempt was to flash a shelly plug, which does not have pins visible - and it was cheaper than a plus 1 pm i have laying around :). But probably more elegant could be a Shelly 1 Plus mini behind a Power Outlet wo load connection.

lucaoliano, 2023-10-16:
If you flash the Shelly with the nuki hub firmware you lose the ability to control any switch/loads and all the other features offered by the stock Shelly firmware.

f-blass, 2024-01-17:
I just installed it on a Shelly Plus 1, worked perfectly with the standard release here. 
The Shelly was freshly sourced (bought from Amazon.de) and it came with the hardware revision 0.1.9. It seems (according to https://github.com/esphome/esphome-devices/blob/main/src/docs/devices/Shelly-Plus-2PM/index.md) that starting with hw revision 0.1.9 also uses dual core ESP32. Apparently Espressif stopped producing the single core variant which Shelly previously used. 
More information on how to identify the dual core version in the document linked.

FBlass, 2024-01-17:
I just installed it on a Shelly Plus 1, worked perfectly with the standard release here.
The Shelly was freshly sourced (bought from Amazon.de) and it came with the hardware revision 0.1.9. It seems (according to https://github.com/esphome/esphome-devices/blob/main/src/docs/devices/Shelly-Plus-2PM/index.md) that starting with hw revision 0.1.9 Shelly also uses dual core ESP32. Apparently Espressif stopped producing the single core variant which Shelly previously used.
More information on how to identify the dual core version in the document linked.

I think the Shelly Plus 1 is a perfect candidate for this as it is small, rather cheap, can be hidden behind an outlet and can be easily flashed with the pinout header on the board (no soldering, only USB-TTL adapter required).

technyon, 2024-01-17:
Thanks for the info.

Also, I think you don't necessarily use the ability to control switch power: Nuki Hub does allow GPIO control, you'd need to know which GPIO turns on and off power.

KingSausage, 2024-01-19:
As I received my UART connector I tried to flash the Shelly 1 PM Plus via the web-interface. After struggling with the pins I managed to get it done. I checked the hw version: it is 0.2.0. I assume, it also uses dual core ESP32.

@technyon : I checked the GPIO section. As far as I understood from the main-page, that in my case I need to set the GPIO to "General output" and need to try each GPIO (high/low) :)? Or does anyone already know which GPIO would "enable"/"disable" the 230V? Than it would be possible to control the power-outlet via mqtt.

f-blass, 2024-01-19:
Yes, there is documentation in the esp-home repo: https://github.com/esphome/esphome-devices/blob/main/src/docs/devices/Shelly-Plus-1/index.md

KingSausage, 2024-01-19:
Ah awesome, in your reference is a reference to the 0.2.1 version with dual mcu. That could be a good hint.
Yes, actually there is not really a need of controlling the power outlet, was just curious if this is possible - although I do not understand how it should work. Nornally you control the relay via the "switch" input. And there are two GPIOs. One tells me to be the output of Relay (GPIO26) , does this mean it controls the output or does it sense. Do you still need the "switch input" (GPIO04). In that case you could hide the shelly behind a light :) and still use switching functionality.

KingSausage, 2024-01-20:
@FBlass Do you have any connection errors with your Nuki Hub in the Shelly? I think I have the same issues like with my esp32. After several hours the bridge goes offline. I didn't install it yet behind the outlet and power it up with USB charger bc I wanted to test it. What distance do you have between controlcenter->shelly->nukilock?
I have like 4m from RPi to hub and another 3m to the lock, almost line of sight only one door (mostly) open.

FBlass, 2024-01-20:
No issues so far. According to my home assistant logs 1 disconnect in the past 4 days, maybe that was even me restarting it because of config changes. Way more stable than my previous dev board.
I have ~3 meters to the Nuki lock (Bluetooth signal strength -57dBm) and ~7 meters to the WiFi router (WiFi signal strength -68dBm). 
Also it is connected to my Nuki Opener.

Maybe try to change the power adapter you're using or just directly wire it to mains voltage. I guess unstable power delivery is often a cause for disconnects.

KingSausage, 2024-01-20:
Thank you for sharing your experience. I think i will consider wiring to the main power source.

KingSausage, 2024-01-26:
With that option enabled it works since a while. That brings the conclusion that the wifi connection is the cause.

For testing purpose I placed the Shelly 1,5 metres away from the router (external power wiring). It is not behind an outlet. I therefore connected the 230V by an external cable :).

PR #341 (2024-04-25): Add C3, S3 and Solo1 support + Build using PlatformIO
-ESP32-S3 build was flashed and tested with success.
-ESP32-C3 and ESP32-Solo1 are untested, but build ok.

Credits for PlatformIO support go to @lucaoliano.
Some changes where required because the bootloader address of the ESP32-S3 and C3 differs from the ESP32.

Closes #199 
Closes #201 
Closes #259 

Leaves CMake support intact.
Artifacts from Github Actions can be used for release zip files and merged binaries in the ZIP files can be used for Webflash.
Could benefit from a release workflow to automate adding binaries to releases/webflas [... cut]
closed 2024-04-25 by technyon