technyon/nuki_hub #200: Drain Nuki Battery

double1968, 2023-06-24:
I have a high battery consumption of my Nuki since I removed the original Hub and put the Esp32 with the latest firmware.  Consumption is 2% each day.
I tried increasing the times but didn't get any results.  is there any solution?
Thanks

technyon, 2023-06-24:
Please post the content of the system information page

double1968, 2023-06-25:
This is my configuration:

```
NUKI Hub version: 8.24
run: true
deviceId: 819686154
deviceIdOp: 819686154
mqttbroker: 192.168.1.100
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: false
ipaddr: 192.168.1.101
ipsub: 255.255.255.0
ipgtw: 192.168.1.2
dnssrv: 192.168.1.2
nwhw: 1
rssipb: -1
hostname: nukihub
nettmout: -1
restdisc: true
rstbcn: -1
lockStInterval: 7200
configInterval: 7200
batInterval: 7200
kpInterval: 3600
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 2.14.5
Lock hardware version: 12.2
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: Yes
Lock has keypad: Yes
Network device: Built-in Wifi
Uptime: 821 minutes
Heap: 93572
Stack watermarks: nw: 6104, nuki: 568, pd: 248
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart

```

technyon, 2023-06-25:
Nothing too obvious in there. You could try extending all "query intervals", so there's less communication with the lock.

To further investigate the issue, logs would be helpful, I think MQTT logs should do in this case: Check "Enable MQTT logging", all log messages will be published into "maintenance/log". You then need to configure your broker to record a history of all messages in that node.

When configured like that, run it for some time and publish the logs here.

double1968, 2023-06-28:
These are mqtt messages querying the Broker.
I don't know if that's correct, messages every 3 seconds.

```
mosquitto_sub -h 192.168.1.100 -u *** -P *** -v  -t "nuki/#"

nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-81
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-78
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-80
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-81
nuki/lock/query/lockstateCommandResult success
nuki/maintenance/uptime 14
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-75
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-78
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-77
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-77
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-80
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-78
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-78
nuki/lock/query/lockstateCommandResult success
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-80
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-78
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-77
nuki/lock/query/lockstateCommandResult success
nuki/maintenance/uptime 15
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-76
nuki/lock/query/lockstateCommandResult success
nuki/presence/devices 54:d2:72:3d:6e:9d;Nuki_1D3D6E9D;-76

```

double1968, 2023-06-28:
Initially yes, then I deactivated it and reassociated it.
how can i reset completely?

technyon, 2023-06-28:
This typically happens when pair with "Register as app" enabled and run NUKI Hub without a NUKI Bridge alongside it. Easiest thing to do is to erase and reflash the ESP, then pair again.

double1968, 2023-06-28:
Ok I try !
Thank you very much

double1968, 2023-06-29:
it works !!!
Thanks

closed 2023-06-29 by double1968