technyon/nuki_hub #202: Template variable error: No first item, sequence was empty. when rendering '{{ (value_json|selectattr('type', 'eq', 'LockAction')|selectattr('action', 'in', ['Lock', 'Unlock', 'Unlatch'])|first).authorizationName }}'

mcervar, 2023-06-29:
Hi

There is a templating issue when using this Nuki esp32 hub with Home Assistant via MQTT

Logger: homeassistant.helpers.template
Source: helpers/template.py:735
First occurred: 13:34:59 (2 occurrences)
Last logged: 13:53:19

Template variable error: No first item, sequence was empty. when rendering '{{ (value_json|selectattr('type', 'eq', 'LockAction')|selectattr('action', 'in', ['Lock', 'Unlock', 'Unlatch'])|first).authorizationName }}'

NukiHub version:  8.24
Firmware version: 3.6.9
Hardware version: 6.11

Home Assistant 2023.6.3
Supervisor 2023.06.4
Operating System 10.3

mcervar, 2023-06-29:
Also, once I locked and unlocked my Nuki lock over the bridge, error is gone ... so I will close the issue

closed 2023-06-29 by mcervar

technyon, 2023-06-29:
Yes this unfortunately happen sometimes, it's a side effect of parsing the access log. There's unfortunately no real better way to do it.