technyon/nuki_hub #203: Compatible ESP32 modules?

goliath888, 2023-07-07:
What modules are compatible?

I have tried with a nodemcu v3 and a lolin d1 mini (v4.0.0) and when I tried to install it via web or with the flash download tool I was not able to do it.

With the web install it says "Your ESP8266 board is not supported."

Is there any list of compatible boards to be sure what I need to buy?

mundschenk-at, 2023-07-07:
ESP8266 is not ESP32

technyon, 2023-07-07:
It's hard to make a list because there are a million different boards with ESP32s out there. I'd recommend the M5Stack Atom Lite:

https://shop.m5stack.com/collections/m5-controllers/products/atom-lite-esp32-development-kit

It comes with an enclose, and has an improved Antenna. Make sure not to buy the -s3 version.

goliath888, 2023-07-08:
> ESP8266 is not ESP32

Both of them are ESP8266 , you are right, thanks. 

> It's hard to make a list because there are a million different boards with ESP32s out there. I'd recommend the M5Stack Atom Lite:
> 
> https://shop.m5stack.com/collections/m5-controllers/products/atom-lite-esp32-development-kit
> 
> It comes with an enclosure, and has an improved Antenna. Make sure not to buy the -s3 version.

Okey, so I am going to buy that one. Thanks.

closed 2023-07-08 by goliath888