technyon/nuki_hub #204: Pair App with Bridge (HUB)

wwwDESIGN-basti, 2023-07-08:
Hey Guys,

is there a way to pair the nuki app with the esp32 nuki hub, to extend the bluetooth range?
What should i do to do this?

technyon, 2023-07-09:
Hi,

No this is not possible, NUKI Hub works independently of the app. In what way do you need to "extend the bluetooth range". Does your ESP have problems communicating with the opener? In that case you have to move the ESP closer to the opener.

platini76, 2023-07-13:
I think he (as me) want to activate option: nuki web to lock and unlock nuki from app also out phone BT range

technyon, 2023-07-23:
Hi,

Sorry for the late answer. The idea is to connect your NUKI device to your smarthome system (Home Assistant, iobroker, FHEM, OpenHAB, ...) via MQTT. The smarthome system then provides a web interface to control the device.

closed 2024-01-02 by technyon