technyon/nuki_hub #205: Minor issue with "diagnostic" sensors for Home Assistant

katiuskt, 2023-07-09:
Using latest 8.25 release with only a Nuki Opener. Everything set by default, just disabled the smartlock and enabled the opener:
[image]

it seems like a couple of (minor) diagnostic sensors in HO
[image]

It looks like the issue came up because of the path of the state_topic:
[image]
[image]

But there are no such paths but "nuki/maintenance/mqttConnectionState" and "nuki/info/nukiHubVersion".
[image]

I've just manually pubished the right paths and diagnostic values are properly shown:
[image]

mundschenk-at, 2023-07-09:
I have noticed this as well, with time the diagnostic sensors from the main node (the lock) are partially exposed on the secondary node (the opener). I think there is a buffer overrun somewhere in the MQTT code, but @technyon has not been able to reproduce this. 

Are you perhaps using TLS over WiFi, @katiuskt?

katiuskt, 2023-07-09:
No TLS used here.

But not sure I get your point. I think what you said is the opposite of what I see (or I misunderstand your point; apologies upfront if that case): What I see is the "home assistant" topics point out towards the "nukiopener" (I guess because the "MQTT NUKI Opener Path" is set by default to "nukiopener").

So there are like two "fathers" in the tree: 
- "nuki" with Rssi information, uptime, if mqtt connection is online, the nuki hub version...
- "nukiopener" with the Opener stuff

Those diagnostic fields for HA are pointing towards "nukiopener" (~/info/nukiHubVersion, for instance) instead towards "nuki"

If "nuki" sensors were exposed (eventually, due to that potential buffer overrun) to the "nukiopener", information would be shown in HA, so it sounds the opposite, isn't it?

mundschenk-at, 2023-07-09:
They should only be under `nuki`, but with time appear under `nukiopener` as well.

katiuskt, 2023-07-09:
Right! The issue is because the state_topic set in the discovery topic for those two sensor is wrong:

For instance, by default the discovery topic for the homeassistant/sensor/XXXX/nuki_hub_version/config is something like this
```
{
  "dev": {
    "ids": [
      "nuki_redacted"
    ],
    "mf": "Nuki",
    "mdl": "Opener",
    "name": "Intercom"
  },
  "~": "nukiopener",
  "name": "Intercom NUKI Hub version",
  "unique_id": "redacted_nuki_hub__version",
  "stat_t": "~/info/nukiHubVersion",
  "ent_cat": "diagnostic",
  "enabled_by_default": true,
  "ic": "mdi:counter"
}
```

Makes that the state_topic point to **nukiopener**/info/nukiHubVersion when the right link should be **nuki**/info/nukiHubVersion

mundschenk-at, 2023-07-09:
You know, I think that is the actual issue, the shorthand `~` being set to the incorrect prefix sometimes. Not a buffer overflow after all.

katiuskt, 2023-07-09:
> You know, I think that is the actual issue, the shorthand `~` being set to the incorrect prefix sometimes. Not a buffer overflow after all.

It seems so! For those sensors not Opener specific but Nuki generic, it seems ~ should be set as "nuki". Unfortunately I don't know how/where to fix that at level code therefore just raised this :-) Hopefully someone more skilled will be able 👍

technyon, 2023-07-11:
Give this binary a try.

[nuki_hub-8.26-pre-1.zip](https://github.com/technyon/nuki_hub/files/12018271/nuki_hub-8.26-pre-1.zip)

katiuskt, 2023-07-11:
Thank you @technyon 
No luck. Updated OK:
[image]

Restarted (also HA restarted) and yet:
[image]

From MQTT Explorer I still saw the "~" pointint towards "nukiopener" (instead "nuki"):
homeassistant/sensor/XXXX/nuki_hub_version/config
```
{
  "dev": {
    "ids": [
      "nuki_redacted"
    ],
    "mf": "Nuki",
    "mdl": "Opener",
    "name": "Intercom"
  },
  "~": "nukiopener",
  "name": "Intercom NUKI Hub version",
  "unique_id": "redacted_nuki_hub__version",
  "stat_t": "~/info/nukiHubVersion",
  "ent_cat": "diagnostic",
  "enabled_by_default": true,
  "ic": "mdi:counter"
}
```

I removed the topic just in case it should need to be recreated but... it is not recreated and no idea how to recreate them automatically to check if doing tha

I've tried to leave blank the "Home Assistant discovery topic" field in "Advanced MQTT and Network Configuration" -> restart -> set that field again to "homeassistant" -> restart and no luck: the removed topics are not recreated

iranl, 2024-02-06:
Almost all MQTT data is published to the respective lock or opener basetopic except for the following list:

- mqtt_topic_wifi_rssi
- mqtt_topic_uptime
- mqtt_topic_freeheapmqtt_topic_restart_reason_fw
- mqtt_topic_restart_reason_esp
- mqtt_topic_info_nuki_hub_version
- mqtt_topic_network_device
- mqtt_topic_mqtt_connection_state

These are only published to the "_maintenancePathPrefix" which seems to follow the value set for preference_mqtt_lock_path (defaults to "nuki")

Of this list only mqtt_topic_info_nuki_hub_version, mqtt_topic_mqtt_connection_state and mqtt_topic_wifi_rssi are part of home assistant discovery topics. Setting these in the HA discovery topics to the preference_mqtt_lock_path value instead of "~" should fix this issue.

Testing and preparing a PR.

PR #292 (2024-02-07): Fix Home Assistant discovery topics
Closes #205
closed 2024-02-07 by technyon