technyon/nuki_hub #206: Nuki Opener options: ledEnabled, buttonEnabled, soundLevel not working

katiuskt, 2023-07-10:
Recent user of this nuki_hub development (a great one, indeed!) and really excited with my new Opener, this, and its HA integration without Nuki Bridge!

Raised this case because although I can see the `nukiopener/configuration/buttonEnabled`, `nukiopener/configuration/ledEnabled` and `nukiopener/configuration/soundLevel` topics they don't seem to work:
- If using App I enable/disable the opener LED, its status is not updated into MQTT. The other way around also happens: If I toggle the HA led_enabled switch, nothing happens neither in the Opener nor the App
- Similarly with soundLevel: no matter at which level I set the sound slider in the app, it isn't synched with MQTT topic
- Finally, regarding the 'buttonEnabled' option I'm not quite sure which are its expected values. In the App I could set different options depending on 1x (singleButtonPressAction) or 2x (doubleButtonPressAction) button press

So, I'm just wondering: Are those topics expected to be working (thus I'm facing a bug) or maybe they are still work in progress and it is the expected behaviour? 

I've seen this api web https://api.nuki.io/#/Smartlock/SmartlockOpenerAdvancedConfigResource_post_post
and those parameters (and others, like enable/disable the doorbell supression, the electricStrikeDelay) seem to be suitable to be modified somehow. But not sure if this development uses that Nuki Web API or another method to communicate with the Opener.

Thank you in advance

technyon, 2023-07-11:
Hi,

Those nodes represent what is referred to as "Configuration" in the API. If you update them from one device (e. g. Smartphone), other devices (e. g. NUKI Hub) don't know that these values have changed. There's no mechanism (unlike for the  lockstate) to notify other devices that there was a configuration change. NUKI Hub polls the configuration in intervals, you can configure the interval under "Advanced NUKI Configuration", "Query interval configuration".

So if you set the interval for example to 30 minutes, in the worst case it can take 30 minutes for NUKI Hubs to update those nodes if you've updated the configuration via the smartphone.

I'm not sure about buttonEnabled. The API documentation only specifies it as a boolean value, on or off, I'd need to further investigate this (see https://developer.nuki.io/page/nuki-opener-api-1/7/). This could also be a copy/paste error, because the documentation has been copied from the smartlock docs, and some things haven't been updated properly.

Making the doorbell suppression configurable via MQTT is actually a good idea, I'll look into it.

katiuskt, 2023-07-11:
Thank you very much for the explanation. Really appreciated!! 🙌

I didn't know the synchronization APP <-> MQTT was based on a polling process. Reduced the "Query interval configuration" to 5 minutes and I'm able to see the changes done in the app (the sound level) reflected in MQTT/HA and viceversa (enable LED through MQTT/APP and eventually it is reflected in the app too).

It will be really appreciated if new functions available in the app (configuration of button, doorbell supression, opening delay...) will be available in the future. If tested will be needed, I can help on that if needed :-)

closed 2024-01-02 by technyon