technyon/nuki_hub #213: Binary state not updated on MQTT

sciorty, 2023-07-27:
Hi and first of all thanks for this amazing project.
What i want to achieve is to control a nuki 3.0 from home assistant.

### General
Ths lock is correctly connected to the mqtt server and to the device. The state is currently locked:
[image]

### MQTT
I did not touch the default mqtt configurations, apart from the restart flag that is actually needed as i lost control of the lock via home assistant after some hours. (Did not investigated too much, but it seemed like https://github.com/technyon/nuki_hub/issues/198):
[image]

### NUKI
Here i just modified the state publishing interval, as i tought that, at least in the early stage, i want the device status to be fastly reported on HA:
[image]

### Problem
The lock state is never updated in HA. I just had the device after a restart thanks to the discovery and did not modify anything apart from the name.
As you can see it just reports an initial "unlocked" state (i think after a restart?) but it never reports the locked states.

### Additional info
- M5 Atom lite
- Latest FW
- Connected through wifi
- Both HA and M5 have static IP
- The M5 is almost 2m away from the locker, behind a thin drywall

alexdelprete, 2023-07-28:
If HA reports no updates it means either that:

1. MQTT broker is not updated.
2. HA is not communicating well with MQTT.

First step: check 1. Use MQTT Explorer, and see if the nuki_hub topic is being updated and if autodiscovery topics under homeassistant main topic are being correctly updated, that is responsibility of Nuki Hub firmware.

If first step is ok, then the problem is between HA and the broker, so you should check on HA community forums or if you can't solve it I'll try my best to help.

Let me know...

eodabas, 2023-07-30:
I'm having the same issue with the exact same setup; M5 Atom Lite, latest FW, connected through wifi and close proximity to the lock.

What I see here is nuki_hub topic in MQTT is not updated when nuki is locked/unlocked within nuki app/bridge, but updated if I lock/unlock through MQTT.

mundschenk-at, 2023-07-30:
Yeah, don't connect Nuki Hub in parallel to the official bridge. That's a limitation of the BLE API.

eodabas, 2023-07-30:
Yes, I am aware of the limitation, but it is documented that I could use the hub and bridge together, with the caveat that either Nuki Hub or Bridge might miss updates. But in this case, the Nuki hub _never_ updated, but the bridge operates normally.

mundschenk-at, 2023-07-30:
Yeah. That's the way it is. When that documentation was written, we didn't know exactly, now we do. You can do it, but you have to query the state frequently because you don't get updates.

eodabas, 2023-07-30:
Oh, I see. I thought that I might test the functionality of the hub first, before completely disconnecting the bridge. I'll try again re-pairing the hub as the bridge, replacing the official bridge then.

sciorty, 2023-09-27:
> If HA reports no updates it means either that:
> 
> 1. MQTT broker is not being updated by Nuki Hub fw.
> 2. HA is not communicating correctly with MQTT.
> 
> First step: check 1. Use MQTT Explorer, and see if the nuki_hub topic is being updated and if autodiscovery topics under homeassistant main topic are being correctly updated, that is responsibility of Nuki Hub firmware.
> 
> If first step is ok, then the problem is between HA and the broker, so you should check on HA community forums or if you can't solve it I'll try my best to help.
> 
> Let me know...

@alexdelprete sorry for getting back to you so late.
Here is a screenshot of the current situation on MQTT Explorer. Hope it's what you asked for.
As you can see, the last message is "Unlocked" but actually it is currently locked

[image]

alexdelprete, 2023-09-27:
> As you can see, the last message is "Unlocked" but actually it is currently locked

So the topic is being updated but it doesn't reflect the real status of the lock?

Are you still using Nuki Bridge? Read comments above: you can't use Nuki Hub together with the official Nuki Bridge.

sciorty, 2023-09-27:
Nope, i don't have any other physical device other than the nuki3.0 and the atom5.
I once tried to do the bridge on the raspberrypi to which home assistant is running (https://github.com/regevbr/RaspiNukiBridge) but i think i correctly removed it and most importantly the nuki is no more paired to it (did a factory reset)

alexdelprete, 2023-09-27:
On the home web page of Nuki Hub, click on the firmware version link, it will open up the detailed status page of Nuki Hub, and post all the contents here.

sciorty, 2023-09-27:
```
System InformationNUKI Hub version: 8.25run: true
deviceId: 846663255
deviceIdOp: 846663255
mqttbroker: 192.168.1.100
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: true
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: nukiopener
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 5
hostname: nukihub
nettmout: -1
restdisc: true
rstbcn: -1
lockStInterval: 20
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.6.9
Lock hardware version: 4.10
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 826 minutes
Heap: 93264
Stack watermarks: nw: 6168, nuki: 568, pd: 244
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
```

i don’t know about the home assistant discovery name, but i correctly have it exposed on HA mqtt integration. Also, it stopped working just today, i’ll leave it like that because that’s another problem i want to fix.. i see it online and i can access it but does not “hear” my home assistant commands like it usually does

alexdelprete, 2023-09-27:
if hassdiscovery is empty, Nuki Hub won't update the MQTT topic HA uses to update/discover the device. So it has to be filled, otherwise it's impossible you see updates in HA about the lock. You are seeing the OLD values, but they are not updated.

BTW: you are not on latest version, that is 8.26.

This is my debug info, so you can make a comparison also regarding the config values, and I would ask @technyon to take a look at it just to know if he sees some evident issue:

```
NUKI Hub version: 8.26
run: true
deviceId: 472593970
deviceIdOp: 472593970
mqttbroker: mqtt.axel.dom
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: true
lockena: true
mqttpath: nuki_hub
openerena: false
mqttoppath: nukiopener
maxkpad: 2
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 10.1.10.164
ipsub: 255.255.254.0
ipgtw: 10.1.10.1
dnssrv: 10.1.10.1
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: 120
restdisc: true
rstbcn: 180
lockStInterval: 900
configInterval: 900
batInterval: 900
kpInterval: 900
kpEnabled: true
accLvl: 0
regAsApp: false
nrRetry: 3
rtryDelay: 300
crdusr: ***
crdpass: ***
pubauth: true
pubdbg: true
prdtimeout: 180
hasmac: true
macb0: -15
macb1: -94
macb2: 12
MQTT connected: Yes
Lock firmware version: 2.15.0
Lock hardware version: 12.2
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: Yes
Lock has keypad: Yes
Network device: Built-in Wifi
Uptime: 554 minutes
Heap: 80952
Stack watermarks: nw: 6072, nuki: 568, pd: 240
Restart reason FW: BLEBeaconWatchdog
Restart reason ESP: ESP_RST_INT_WDT: Reset (software or hardware) due to interrupt watchdog
```

sciorty, 2023-09-27:
Thanks! That was it! I just put “homeassistant” and it worked.
I just rushed because it appeared to work without configuring it, now the state is reported

closed 2023-09-27 by sciorty

sciorty, 2023-09-28:
Just a question: will the lock state be reported via MQTT at every state change or it does it only at the interval specified in the web page field? I want to save as much battery as possible and i don't care to poll the lock avery minute or so, i just want the changes to be reported

alexdelprete, 2023-09-28:
You're confusing things: reporting to MQTT is done by the Atom, so it doesn't consume any battery.

There are 2 things: polling by the Atom, configurable (I have it at 15m), and also when the lock state changes, the lock itself pushes the change to Atom via BLE, the Atom then pushes the change to MQTT.

This is my configuration for the polling periods, experiment with those values for your setup:

[image]