technyon/nuki_hub #214: Nuki - Trigger Switching

CAVACO-PT, 2023-07-27:
Hi all , iv tried to figure this , out been checking everywhere , but  my lock started to switch triggers auto , some during the day it keeps i self in autolock , and locks after a whille , and at night opens it selft and stays the whole night open ...

Can u please help me in  understanding where i can change this i want the other way arround autolock during the night , and during the day , leaves it to manual 

[image]

Thank you for the help

technyon, 2023-07-28:
Hi,

although you've closed the issue, I can give you an advice: If you're using the NUKI Widget on your phone, remove it. It rarely but randomly opens the door.

CAVACO-PT, 2023-07-28:
Hi  , i tought it went away ,but its closing the door no during the day .... so its not solved , i can try to remove the widget to test !!!

technyon, 2024-01-02:
Closing due to inactivity

closed 2024-01-02 by technyon