technyon/nuki_hub #217: MQTT discovery naming issues warnings in Home Assistant 2023.8

Mincka, 2023-08-05:
TL;DR: The name of the device should not be prefixed in the discovery info.

If I understand correctly.

BEFORE:
`"name": "Porte Cuisine LED enabled",`
```
{
  "dev": {
    "ids": [
      "nuki_310b3855"
    ],
    "mf": "Nuki",
    "mdl": "SmartLock",
    "name": "Porte Cuisine"
  },
  "~": "nuki_cuisine",
  "name": "Porte Cuisine LED enabled",
  "unique_id": "310b3855_led_enabled",
  "stat_t": "~/configuration/ledEnabled",
  "ent_cat": "config",
  "cmd_t": "~/configuration/ledEnabled",
  "ic": "mdi:led-variant-on",
  "pl_on": "1",
  "pl_off": "0",
  "state_on": "1",
  "state_off": "0"
}
```

AFTER:
`"name": "LED enabled",`
```
{
  "dev": {
    "ids": [
      "nuki_310b3855"
    ],
    "mf": "Nuki",
    "mdl": "SmartLock",
    "name": "Porte Cuisine"
  },
  "~": "nuki_cuisine",
  "name": "LED enabled",
  "unique_id": "310b3855_led_enabled",
  "stat_t": "~/configuration/ledEnabled",
  "ent_cat": "config",
  "cmd_t": "~/configuration/ledEnabled",
  "ic": "mdi:led-variant-on",
  "pl_on": "1",
  "pl_off": "0",
  "state_on": "1",
  "state_off": "0"
}
```

> The change states that MQTT should not provide the device name in the entity names that are attached to the device. So if your upstream MQTT provider is supplying a device name inside the entity name, they should stop providing that information. They should only provide the device name to the device, not the entity.

More info:
https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

[image]
*Extract of warnings in HA 2023.8*

**This is not a breaking change and the entity name will not change.**

[image]

That's because they plan to use the device name as a prefix for *the name* the entities starting in 2024.4
https://www.home-assistant.io/blog/2023/08/02/release-20238/

> This means any MQTT entity which is part of a device will [automatically have its friendly_name attribute prefixed with the device name](https://developers.home-assistant.io/docs/core/entity/#has_entity_name-true-mandatory-for-new-integrations).

technyon, 2023-08-05:
ok, should be easy to fix

alkertrazz, 2023-08-05:
> ok, should be easy to fix

Do anyone no how to fix it? Because unfortunately my Nuki doesn't react in Home Assistant.

technyon, 2023-08-12:
Please check the attached binary.

[nuki_hub-8.26-pre-2.zip](https://github.com/technyon/nuki_hub/files/12327214/nuki_hub-8.26-pre-2.zip)

jressel01, 2023-08-13:
i test 8.26 pre 2. the Sensor and switches are ok. Only the lock.xxxxx are left with this problem

technyon, 2023-08-13:
For the lock, the name is the name configured in the NUKI App. What is suggested to be put here instead? "lock" for the lock, and "opener" for the opener?

jressel01, 2023-08-13:
i think you can remove the name near unique_id. because You have it in the first Part at all othe devices. I do the same at my wBecathermoastat. it take the name from Device

OnFreund, 2023-08-14:
Yes, the main entity of the device - the lock in this case - should not have an explicit name (not sure if for MQTT that means setting it to `null` or simply omitting it). It will take the name from the device name.

technyon, 2023-08-14:
Here you go, name is completely removed.

[nuki_hub-8.26-pre-3.zip](https://github.com/technyon/nuki_hub/files/12337401/nuki_hub-8.26-pre-3.zip)

jressel01, 2023-08-14:
all is fine. No warning. thx

Mincka, 2023-08-17:
Thank you for that! Just a quick question, the update 8.26 does not seem to include the latest commit with the removal of the name for the device itself.
In this [specific commit](https://github.com/technyon/nuki_hub/commit/33051682e8dd0dbbbf2718c3d28374cc0dc6bfd3), the version is also updated to 8.27.  Does it mean it's indeed not in the latest release version (8.26)?

Mincka, 2023-08-17:
In fact,

technyon, 2023-08-22:
So this would be correct?

[nuki_hub-8.27-pre-1.zip](https://github.com/technyon/nuki_hub/files/12411110/nuki_hub-8.27-pre-1.zip)

Mincka, 2023-08-24:
Sorry if it was not clear, but that's the opposite, keep the name of device, like it's done for all the entries, but for the lock, set the entity name to "None".
[image]

Because, with this update, the device is now called "None".
[image]

Thank you for the updates.

mundschenk-at, 2023-08-25:
I don't think `"None"` as a string value is correct anyway. It should be unquoted `null` in the payload.

Arn0uDz, 2023-09-08:
Any updates on this? Getting the MQTT error in my logs.

@Mincka could you re-open this issue?

technyon, 2023-09-08:
If "null" is agreed upon to be the correct value, I can change it.

Arn0uDz, 2023-09-09:
@Mincka @jressel01 @mundschenk-at agreed?

mundschenk-at, 2023-09-09:
It's plain `null` though, not a quoted string value. I think you also can just leave out the key with the same result, though.

Arn0uDz, 2023-09-10:
> If "null" is agreed upon to be the correct value, I can change it.

Can you change it to null without quotation marks?

Mincka, 2023-09-10:
> @Mincka could you re-open this issue?

No sorry, I cannot reopen this issue because I'm not the one who closed it.

jressel01, 2023-09-11:
i change the file and make a Pullup Request.[https://github.com/technyon/nuki_hub/pull/225](url)

technyon, 2023-09-11:
I've built a binary upon those changes, please let me know if works like this.

[nuki_hub-8.27-pre-1.zip](https://github.com/technyon/nuki_hub/files/12575390/nuki_hub-8.27-pre-1.zip)

Arn0uDz, 2023-09-12:
@technyon Thanks. Just tried it and solved the MQTT warnings in HA. The Nuki lock still has the correct name in HA.

Mincka, 2023-10-10:
Hello @technyon, could you release the version with this fix? Thank you!

technyon, 2023-10-11:
Yes, I'd like to get feedback on issue #227 so I can release both together

technyon, 2023-10-15:
Fixed in release 8.27