technyon/nuki_hub #218: Manual unlock/lock not reporting

deangreen4, 2023-08-06:
Hi,
am running version 8.25 of nuki bridge with an ultion lock which uses the nuki 3.0. All is well if the lock is unlocked with the nuki app, clicking the button or using home assistant but if you just turn the knob on the nuki, or use a key from the outside, this does not cause a state change on the bridge. The nuki app shows this state change. Any ideas?

Thanks

technyon, 2023-08-06:
I don't have too much information about that lock, but once a state change happens, it should signal it via a flag in the ble beacons. My guess would be that for some reason it doesn't set that flag.

deangreen4, 2023-08-09:
Oddly, without me doing anything else - its started working perfectly!
Thanks for all your hard work on this code

closed 2023-08-11 by deangreen4