technyon/nuki_hub #223: the battery discharges quickly

amastrogiacomo1968, 2023-09-06:
Good morning, I've noticed that in recent months the battery is discharging more quickly. From Sunday until today, it went from 100% to 90% :-( . Are there any settings I can check that might be increasing the consumption? Of course, the use of the lock is more or less the same. Thanks in advance.

technyon, 2023-09-06:
First, post the content of the system information page.

You can try adjusting the query intervals as a first measure. The longer the interval, the shorter the battery drain.

amastrogiacomo1968, 2023-09-07:
this is system information:

NUKI Hub version: 8.26
run: true
deviceId: 4087870833
deviceIdOp: 4087870833
mqttbroker: 192.168.1.18
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: false
ipaddr: 192.168.1.234
ipsub: 255.255.255.0
ipgtw: 192.168.1.1
dnssrv: 192.168.1.1
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: 600
restdisc: true
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: true
nrRetry: 0
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 2.14.5
Lock hardware version: 12.2
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: Yes
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 2 minutes
Heap: 90300
Stack watermarks: nw: 6184, nuki: 568, pd: 276
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_BROWNOUT: Brownout reset (software or hardware)

is there any particular query intervals that I can adjust better?
Thank you

technyon, 2023-09-07:
Are you using a NUKI Bridge in addition to NUKI Hub?

technyon, 2023-09-07:
Please:

- Disable "Register as app"
- unpair NUKI Hub
- then re-pair

Was there any reason you've enabled this setting?

amastrogiacomo1968, 2023-09-07:
No particular reason :-)
I did what you told me, by the way I saw from the app that there was a fw update I did.
I'll try a few days and let you know.
For now thanks

technyon, 2023-09-07:
I'm very confident this will solve your problem.

amastrogiacomo1968, 2023-09-12:
I'd say the problem is solved. Thanks for your help

closed 2023-09-12 by amastrogiacomo1968