technyon/nuki_hub #224: Template variable error: No first item, sequence was empty. when rendering '{{ (value_json|selectattr('type', 'eq', 'LockAction')|selectattr('action', 'in', ['Lock', 'Unlock', 'Unlatch'])|first).authorizationName }}'

mcervar, 2023-09-07:
Hi

On each restart of Home Assistant I am getting this error:

Template variable error: No first item, sequence was empty. when rendering '{{ (value_json|selectattr('type', 'eq', 'LockAction')|selectattr('action', 'in', ['Lock', 'Unlock', 'Unlatch'])|first).authorizationName }}'

Its regarding the templating, is it possible to fix it somehow?

technyon, 2023-09-07:
Post the content of the system information page please.

mcervar, 2023-09-07:
Sure

Home Assistant 2023.8.4
Supervisor 2023.08.3
Operating System 10.5
Frontend 20230802.1 - latest

Hostname | nukihub
MQTT Connected | Yes
NUKI Lock paired | Yes
NUKI Lock state | unlocked
Firmware | 8.26

mcervar, 2023-09-08:
ahaaa okks :D

> NUKI Hub version: 8.26
run: true
deviceId: 363800419
deviceIdOp: 363800419
mqttbroker: 192.168.0.105
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: 60
restdisc: true
rstbcn: 60
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: true
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.6.9
Lock hardware version: 6.11
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: Yes
Network device: Built-in Wifi
Uptime: 1205 minutes
Heap: 92520
Stack watermarks: nw: 6136, nuki: 472, pd: 236
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.

mundschenk-at, 2023-09-24:
I think we talked about it this some time back, @technyon. The template sequence I devised does not properly deal with an empty topic. I can look into it.

mcervar, 2023-10-06:
Any update on this? :D

technyon, 2023-10-07:
Hi. I'm quite busy at the moment, but I'll look into it.

mundschenk-at, 2023-10-08:
Hi @technyon, it should work correctly when you change the value template to

```jinja2
{{ (value_json|selectattr('type', 'eq', 'LockAction')|selectattr('action', 'in', ['Lock', 'Unlock', 'Unlatch'])|first|default).authorizationName|default }}
```

technyon, 2023-10-09:
Hi,

I've applied your changes. Please test the binary I posted in issue #227, the fix is in there

mcervar, 2023-10-09:
Hi

I just tried it, I dont see error in HA logs anymore, thx!

closed 2023-12-27 by technyon