technyon/nuki_hub #227: Change Lock entity to Unavailable when MQTT is disconnected

leranp, 2023-09-12:
Hi,
Love this component, but there is something that i think that needs to be changed, when the MQTT is disconnected for some reason, the lock entity still shows status that can be worng.
In all of my MQTT base devices, when the MQTT is not connected, the device goes to Unavailable State

technyon, 2023-09-12:
It wouldn't make sense to change the lock state if MQTT is disconnected. The lock doesn't necessarily change state just because MQTT is disconnected.

What you want to monitor is the "maintenance/mqttConnectionState" node, which reflects wether or not the ESP is connected ... which reminds me I'll have to add this to the readme.

mundschenk-at, 2023-09-12:
I disagree, in HA that absolutely would make sense. When the ESP is disconnected from MQTT, the lock state is `unavailable`. Adding it to auto-discovery should be pretty straightforward, I can try to come up with something.

technyon, 2023-09-13:
ok, understood. However @leranp didn't mention this is HA related.

leranp, 2023-09-13:
Yes, i am talking about HA

technyon, 2023-09-13:
@mundschenk-at Let's update the auto discovery then. We can release this together with the fix for the auto discovery warnings.

mundschenk-at, 2023-09-20:
Sorry for the delay, I'll do the write-up this weekend.

technyon, 2023-09-21:
@mundschenk-at ok

mundschenk-at, 2023-09-23:
So in YAML you'd need to add this to auto-discovery:
```yaml
      availability:
        - topic: <prefix>/maintenance/mqttConnectionState
```

No need to set the payloads, as they default to `online` and `offline` anyway. So for the JSON payload, it would be:

```json
"avty": [{"t": "~/maintenance/mqttConnectionState"}]
```

I am using the `availability` syntax here because it would allow us to add several availability topics, e.g. not only for the MQTT connection but also for the BLE connection - the state of the lock is only accurate if there was recent-ish BLE connection and MQTT is connected. You could set such a topic to `offline` when a BLE restart is necessary. That would make such interruptions visible in the HA logbook.

mundschenk-at, 2023-09-23:
Anyway, it was just a thought, probably too niche a use-case. The MQTT availability topic should probably be added to all auto-discovery entities except the `MQTT connected state`.

technyon, 2023-09-25:
I've added availability to the json as suggested. Please check and confirm that it works. So far only for the lock, I'd added it to the other entities if it's confirmed it works for the lock.

[nuki_hub-8.27-pre-2.zip](https://github.com/technyon/nuki_hub/files/12710960/nuki_hub-8.27-pre-2.zip)

mundschenk-at, 2023-09-25:
I tried it, works fine for the lock, but It breaks the opener because the connection state is not available under the opener prefix. Not sure if you can separate them with inheritance? 

I do have a suggestion how to handle this (with my long-promised write-up of a better auto-discovery device split), but that probably needs some discussion.

mundschenk-at, 2023-09-27:
This would be the quick fix, but it is kind of dirty. Or you could simply reference the `mqttConnectionState` under the lock prefix.

IMHO nicer would be to move all the Nuki Hub/ESP-specific topics to their own HA "hub" device (and maybe MQTT topic, but that would be a breaking change of course) and have the Lock and Opener be connected devices to that "hub". (I'd have to look into it whether we would then still need the separate availability topic in the Lock and Opener device definitions, maybe we do.)

technyon, 2023-09-29:
All paths are relative right? Should be easy enough to point to the lock maintenance topic.

technyon, 2023-10-09:
Please check if it works like this.

[nuki_hub-8.27-pre-3.zip](https://github.com/technyon/nuki_hub/files/12846294/nuki_hub-8.27-pre-3.zip)

mcervar, 2023-10-09:
Hi 

I just tried it, I dont see error in HA logs anymore, thx!

leranp, 2023-10-09:
> Please check if it works like this.
> 
> [nuki_hub-8.27-pre-3.zip](https://github.com/technyon/nuki_hub/files/12846294/nuki_hub-8.27-pre-3.zip)
> 
> 

Looks fine for me

technyon, 2023-10-09:
@mundschenk-at Does it fix the issue with the opener?

mundschenk-at, 2023-10-09:
> @mundschenk-at Does it fix the issue with the opener?

Unfortunately, no. I missed your question about paths being relative, to which the answer would be no (or rather: MQTT topics are not paths that can be navigated). So 
```json
"availability": {
  "topic": "nukiopener/../nuki/maintenance/mqttConnectionState"
}
```
is incorrect, it would need to be
```json
"availability": {
  "topic": "nuki/maintenance/mqttConnectionState"
}
```

mundschenk-at, 2023-10-11:
Sorry I wasn't more explicit yesterday, the new version works fine.

technyon, 2023-10-15:
Check release 8.27

bcutter, 2023-10-20:
OK I updated to 8.27 some days ago (same day it became available). Now that I restarted HA for the first time since then, all my locks are shown as `unavailable`, even they are reachable and at the same time the MQTT connected entities show `on`.

Not sure what screwed things up - I've NEVER ever seen the lock entities in `unavailable` state before the 8.27 update.
Will try to downgrade one Nuki hub to 8.26 for now and see if that changes things.

[image]

[image]

[image]

technyon, 2023-10-23:
What path did you configure for the lock? And could you share the content of the HA discovery topic for the lock?

bcutter, 2023-10-24:
Could you please guide me precisely on how to get that information?

Afaik I never changed/customized things so everything _should_ be default.

technyon, 2023-10-24:
As a start, maybe post the content of the system information page.

Which MQTT broker are you using?

mundschenk-at, 2023-10-24:
> Could you please guide me precisely on how to get that information?

In HA: Settings > Devices & Services > MQTT > Your lock device > 3 dots > Download diagnostics

bcutter, 2023-10-24:
System information page (masked)
```
NUKI Hub version: 8.26
run: true
deviceId: XXXXXXXXXXX
deviceIdOp: XXXXXXXXXXX
mqttbroker: xxx.xxx.xxx.xxx
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki_doorone
openerena: false
mqttoppath: 
maxkpad: 5
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: xxx.xxx.xxx.xxx
ipsub: 255.255.255.0
ipgtw: xxx.xxx.xxx.xxx
dnssrv: xxx.xxx.xxx.xxx
nwhw: 1
rssipb: -1
hostname: ESP-Nuki-Hub-DoorOne
nettmout: -1
restdisc: true
rstbcn: 60
lockStInterval: 3600
configInterval: 3600
batInterval: 7200
kpInterval: 1800
kpEnabled: true
accLvl: 
regAsApp: false
nrRetry: 5
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: true
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.7.7
Lock hardware version: 5.11
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: Yes
Network device: Built-in Wifi
Uptime: 1088 minutes
Heap: 85424
Stack watermarks: nw: 6104, nuki: 360, pd: 248
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

As the MQTT -> lock -> burger menu -> diagnostics output is very very chatty, here's what I think you might have asked for:
```
        {
          "entity_id": "lock.schloss_doorone",
          "subscriptions": [
            {
              "topic": "nuki_doorone/lock/binaryState",
              "messages": [
               ...................................
                {
                  "payload": "locked",
                  "qos": 0,
                  "retain": 0,
                  "time": "2023-10-24T06:18:06.074445+00:00",
                  "topic": "nuki_doorone/lock/binaryState"
                },
                {
                  "payload": "unlocked",
                  "qos": 0,
                  "retain": 0,
                  "time": "2023-10-24T06:29:16.264524+00:00",
                  "topic": "nuki_doorone/lock/binaryState"
                }
              ]
            }
          ],
```

Hope that helps.

mundschenk-at, 2023-10-24:
No, the discovery topic should be something like `homeassistant/lock/<lockid>/<entityname>/config` (under the key `discovery_data`). However, I've just checked mine and there is some corruption going on. The lock entity shows an incorrect discover topic (`~/maintenance/mqttConnectionState`). However, this looks like a display issue with the HA MQTT integration. The actual discovery topic still exists on my broker.

bcutter, 2023-10-24:
One picture maybe tells more than 1.000 words. Here's once again the current status:

[image]

technyon, 2023-10-25:
Could you post the whole content of that topic?

bcutter, 2023-10-25:
I die so already for one of the two locks, see at the bottom of this post https://github.com/technyon/nuki_hub/issues/227#issuecomment-1777984588

bcutter, 2023-10-25:
Yes everything I provided is from Nuki Hub 8.26. I don't see "avty". Where exactly should I look at?

Yes, this is as expected:
[image]

Existing and updating regularly. On 8.26 (⚠️).

Not sure what else I could provide...

technyon, 2023-10-26:
That's what has been added with 8.27, so you won't see it with 8.26

bcutter, 2023-10-26:
OK. So I guess I can't add something helpful, do I?

technyon, 2023-10-26:
Well... you could update to 8.27 and copy/paste what's there, then go back to 8.26

bcutter, 2023-10-26:
Updated one Nuki Hub again to 8.27. autodiscovery topic (`config`)...
[image]
...now has:
[image]

And that one still exists of course:
[image]

So I assume the `avty` is not available or not correct or something else during HA restart. Otherwise I can't imagine why the lock entity renders unavailable (which for the records it did not during the short update 8.26-8.27 as I did not restart HA).

Let's compare auto discovery topic:
8.26:
```
{"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock"},"~":"nuki_doortwo","name":"Door Two","unique_id":"XXXXXXXX_lock","cmd_t":"~/lock/action","pl_lock":"lock","pl_unlk":"unlock","pl_open":"unlatch","stat_t":"~/lock/binaryState","stat_locked":"locked","stat_unlocked":"unlocked","opt":"false"}
```
8.27:
```
{"dev":{"ids":["nuki_XXXXXXXX"],"mf":"Nuki","mdl":"SmartLock","name":"Door Two"},"~":"nuki_doortwo","name":null,"unique_id":"XXXXXXXX_lock","cmd_t":"~/lock/action","avty":{"t":"~/maintenance/mqttConnectionState"},"pl_lock":"lock","pl_unlk":"unlock","pl_open":"unlatch","stat_t":"~/lock/binaryState","stat_locked":"locked","stat_unlocked":"unlocked","opt":"false"}
```

Comparing both I only discovered that
a) `name` exists two times now with 8.27,
b) first `name` has another position but is correct and
c) second `name` is on the same place as with 8.26 but is `null` (is that suspicious or relevant at all?)

mundschenk-at, 2023-10-26:
That's on purpose due to how recent HA versions handle device vs. entity names.

bcutter, 2023-10-26:
OK. So could that be a definite issue, running an older HA version?

bcutter, 2023-10-26:
> Normally `<prefix>/maintenance/mqttConnectionState` will only change to `offline` if the hub is disconnected from the MQTT broker (by pulling power for example). A reboot of the ESP will not change the topic to `offline` (probably because it is too fast for the session timeout).

Yes, but as initially described here https://github.com/technyon/nuki_hub/issues/227#issuecomment-1773373364 with

> all my locks are shown as `unavailable`, even they are reachable and at the same time the MQTT connected entities show `on`.

**that's not the issue. MQTT connection state completely fine, but lock entities `unavailable`.**

> Which version are you running?

Im running HA Core 2023.3.6 at the moment.

So now it's up to you to figure out
a) why this happens and
b) if/why only for me (what's the update pace, how often has 8.27 been downloaded, how many of those use HA etc.)

I don't see what else I could provide. 8.26 everything's running fine, 8.27 screws up lock entities once HA restarts. No idea what you changed in detail in the 8.27 code.

technyon, 2023-10-27:
From the issues received here, I'd say the majority of users use HA. The code that changed in details doesn't matter all that much, what's important is what's in MQTT. "avty" was added so HA knows when the ESP goes offline. Why it doesn't work in your case is somewhat of a mystery, since everything looks fine.

We can try to figure out what's different in your configuration. What MQTT broker are you using, maybe a specific broker causes problems. Also you have an underscore in the path ... that shouldn't matter but then it could. Maybe as a test change the path to not use an underscore.

bcutter, 2023-10-27:
I'm just using the default Mosquitto broker shipped as addon for HA (https://github.com/home-assistant/addons/tree/master/mosquitto).

Well, instead of ruining my productive system and making my other houshold members sharpen their knifes, I hope you experts have the possibility of a test system. So you could just add a (very very VERY likely unproblematic `_`) to one of your locks and give it a go.

mundschenk-at, 2023-10-27:
Well, you might also want to update your HA version to something more recent. That should not be the issue, but as is, there are a lot of variables here. You can't expect people here to reproduce your exact system.

bcutter, 2023-10-27:
> Well, you might also want to update your HA version to something more recent. That should not be the issue, but as is, there are a lot of variables here.

Yes, I will update HA once there's the needed time. I also don't think this will make a difference as I checked all breaking changes in all monthly release notes between 2023.3 and 2023.10 - nothing in MQTT which could have an impact in this way it renders (lock) entities unavailable as false positive.

> You can't expect people here to reproduce your exact system.

No of course not. But one can the pros have more test possibilities and rule out low-level assumptions like a certain character in a name :-)

bcutter, 2023-11-03:
Small note: found something in HA 2023.10 release notes which sounds kinda interesting:

https://www.home-assistant.io/blog/2023/10/04/release-202310/#breaking-changes

> An MQTT lock with a configured state topic will initialize with state unknown instead of state unlocked unless the lock is set to optimistic mode.

> MQTT locks set to optimistic mode will still be initialized with the unlocked state. You should check if your automations are affected.

While I'm not 100 % sure what the detailed impact of this change on lock entities in HA is, it's impossible my issue is caused by this - as I'm still few release cycles behind that 2023.10 update. Maybe you want do take a look at this change anyway in case you would need to configure something.

technyon, 2023-11-04:
I'm not really sure how to reproduce this. Maybe updating your system first would make sense so we can eliminate at leasta one variable. Other than that, we could think about a way to disable publishing avty.

bcutter, 2023-11-06:
Going through the updates, but it takes some time, probably weeks.
2023.4 made no difference so far (and tbh I doubt it will be fixed with a HA update).

So annoying when updates break (core) things... 8.26 was running so smooth and stable. I get the "want have" factor for having lock entities unavailable depending on the MQTT connection state, but well... not working.

**Maybe we can have this avty thing as _config option_?** So being on as default this way everyone could simply decide on his/her own.

mundschenk-at, 2023-11-07:
Well, if you mean @alexdelprete and me, yes, we are using EMQX. Not sure about everyone else. The broker should not make any difference here though, unless it is somehow possible to disable LWT in Mosquitto. 

@bcutter Please look at your HA log to see if there is any issue with MQTT discovery messages not being parsed - the component is  quite chatty. If not, try to enable debug logging for the `mqtt` component.

alexdelprete, 2023-11-07:
> I wonder if it could be somehow related to the broker. I think everyone else uses EMQX ... is this correct?

It's not broker related. It's an issue with @bcutter HA not parsing tha `availability` topic correctly for some unknown reason. Also, I would use `availability_topic` vs `availability`, since we only use one topic and not multiple topics to determine the availability status. 

```json
"avty_t": "~/maintenance/mqttConnectionState"
```

@bcutter can you try an older version of NH, pre 8.27, to exclude that the `availability` topic (introduced in 8.27) is an issue in your case?

alexdelprete, 2023-11-07:
> While I'm not 100 % sure what the detailed impact of this change on lock entities in HA is

That breaking change simply meanst that on HA startup, MQTT Locks have an unknown state, until the state is correctly determined by the relevant MQTT topic. Basically HA doesn't assume a default state, unless optimistic mode is configured.

bcutter, 2023-11-07:
Yes indeed, 8.26 works without any issues - because it does not have the avty set yet.

So many different approaches here (plenty of activity, appreciate that), not sure what to do / check next on my side. If you experts isolated the core issue please let me know what I can contribute.

I'm kinda sick and tired of updating NH to 8.27 and restarting HA just to see all my lock entities fail (unavailable) again, I've gone through that process several times lately. If I really need to do that one more time I want to know exactly what to check/do/provide you here.

alexdelprete, 2023-11-07:
> If you experts isolated the core issue please let me know what I can contribute.

The core issue seems to be specific to your setup, I haven't seen other reports after 8.27 with your same issue. Please correct me if I missed any. My installation works perfectly fine with 8.27 and my lock doesn't go unavailable.

> If I really need to do that one more time I want to know exactly what to check/do/provide you here.

I suggested to change `avty` to `avty_t`, so when dev provides that test version you might want to test it. What version of HA are you running now? Can you also describe where HA and MQTT broker are running? RPi, VM, LXC, etc. Thanks.

alexdelprete, 2023-11-07:
@bcutter can you tell me if the `mqttConnectionState` topic has the retain flag in the broker? I'm using MQTT Explorer to navigate the topics in the broker here.

[image]

technyon, 2023-11-07:
> I suggested to change `avty` to `avty_t`, so when dev provides that test version you might want to test it. What version of HA are you running now? Can you also describe where HA and MQTT broker are running? RPi, VM, LXC, etc. Thanks.

I think you can test it without a new binary. Just start NUKI Hub and let it publish the discovery topic, then manually change avty to avty_t using some MQTT client. NUKI Hub will only overwrite the discovery topics on boot.

mundschenk-at, 2023-11-12:
OK, after sleeping over it, I now know why @bcutter's lock entities are unavailable with 8.27 - it's got nothing whatsoever to do with the availability topic, but simply a the same as https://github.com/technyon/nuki_hub/issues/246: The new empty `name` in the auto-discovery topic is invalid before Home Assistant 2023.8.0.

alexdelprete, 2023-11-12:
> The new empty `name` in the auto-discovery topic is invalid before Home Assistant 2023.8.0.

So PR #225 causes the entities to be unavailable in HA < 2023.8.0 ?

If confirmed, we should put a note in readme of the repo, specifying HA 2023.8.0 as the minimum supported version. Maybe a pre-requirements paragraph.

Also, when these breaking changes are implemented, they should be highlighted as breaking in release notes so that users know they need to test it first.

bcutter, 2023-11-12:
Great finding, congratulations 👏🏼 👍🏼

Why is it empty at all? Why not setting it?

That way it would resolve any dependency of a HA version, no breaking change notice needed (which is now too late anyway for 8.27) etc.

alexdelprete, 2023-11-12:
> Why is it empty at all? Why not setting it?

It's an HA breaking change: read #217 

[image]

> That way it would resolve any dependency of a HA version, no breaking change notice needed

That's not true, if HA changes something components/integrations need to adapt to those changes, hence minimum supported versions of HA need to be specified.

> which is now too late anyway for 8.27

not true either, there might be users with old HA versions that will install 8.27 and will read the instructions, so minimum requirements should always be mentioned since there are dependencies.

mundschenk-at, 2023-11-12:
@alexdelprete: Right, this should in the README but we didn't really know until now. @technyon doesn't use HA at all and I'm always on the latest version (since I find quickly adapting my configuration to any breaking changes means much fewer issues in the long run). Unfortunately, the Powers That Be at HA didn't highlight the fact that this auto-discovery change would prevent it from working with older versions of HA. 

As a workaround, we could add `name: Lock` for the `lock` entities instead, which would result in the entity name  `lock.your_device_lock` and a friendly name of `Your Device Lock`, but personally I think a simple note in the README should suffice at this point.

@bcutter:
> Why is it empty at all? Why not setting it?
> 
> That way it would resolve any dependency of a HA version, no breaking change notice needed (which is now too late anyway for 8.27) etc.

Because that's the way it is intended: https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/

If you want to stay on something older than 2023.8.0, you can always configure the `lock` entity yourself using YAML (even if you use auto-discovery for the other entities).

alexdelprete, 2023-11-12:
> Right, this should in the README but we didn't really know until now. @technyon doesn't use HA at all and I'm always on the latest version

we adopted a change due to HA recommendations, so we should mention that as a breaking change. we knew it when we implemented it. We were relying on the new 2023.8.0 way of working, to avoid warnings. Doesn't matter if @technyon uses HA or not, we knew the impact when we adopted those changes.

The general readme should always contain pre-requirements, and in the release notes breaking changes (such as a pre-requirement change) should be highlighted properly.

It's a general advice for the future. :)

mundschenk-at, 2023-11-12:
Well, at oeast I didn't know that the auto-discovery message would simply fail to parse on HA < 2023.8.0 (instead of using some sane default) 🤷

alexdelprete, 2023-11-12:
> at least I didn't know that the auto-discovery message would simply fail to parse on HA < 2023.8.0

I understand. But in general, if you adopt a change based on HA version, you have to mention that as an important change and mention it. It's no big deal...but in the future we should not rely on intuitions or memory, we'd waste a lot of time for these issues. :)

[16 more replies]

closed 2023-11-13 by technyon