technyon/nuki_hub #231: The unlatch function does not appear anywhere.

a9965, 2023-09-20:
Where can I find the unlatch function?

mundschenk-at, 2023-09-20:
If you mean in Home Assistant, it is `lock.open`.

a9965, 2023-09-20:
I do not know what you mean. My entity is called "entrada" and I have tried what you mentioned but it doesn't work.

mundschenk-at, 2023-09-20:
The service to call is named `lock.open` (with your specific `lock` entity as the target).

a9965, 2023-09-20:
Is the code I showed you above what you mean? If not, I don't know how to do it. Thank you.

mundschenk-at, 2023-09-20:
This looks like part of a dashboard configuration, I don't know what exactly your are trying to do there. I suggest you test your service calls through the Developer Tools.

SergioWarriorPeru, 2023-09-22:
> Is the code I showed you above what you mean? If not, I don't know how to do it. Thank you.

I use node red and I have configured in this way
[image]

iranl, 2024-02-06:
An auto discovered Unlatch button for Home Assistant was added in #285 en will be part of the next release. 

Should also close this issue.

technyon, 2024-02-06:
Correct. Closing please wait for the next release.

closed 2024-02-06 by technyon