technyon/nuki_hub #232: [Suggestion/Question] Have it been considered to use NukiHub as a docker container?

anthonws, 2023-09-24:
I'm looking at the possibility of using NukiHub as a docker container in a NUC, that packages both WiFi + Bluetooth and is running my HA docker installation. This would save the need to yet again support another device (albeit inexpensive).
There are other projects like NukiPyBridge, RaspiNuki, etc. But all of them have several gaps/limitations and are not as mature as NukiHub is (congrats on the amazing project!).

Long story short, has anyone used NukiHub in a Linux host or container environment? I'm researching also the ability to emulate ESP-IDF via QEMU. IF anyone has some pointers, please feel free to share.

Cheers,
anthonws.

mundschenk-at, 2023-09-24:
Not sure if ESP emulation is a thing (I doubt it), but there won't ever be a way to run Nuki Hub in a Docker container. It's not a Linux software and not easily portable to such an environment.

technyon, 2023-09-25:
As already said, I doubt this is possible. Even if you could emulate an ESP32, it would somehow need to interface with the hosts bluetooth and wifi hardware, which would need some tricky wrapper to translate low-level ESP bluetooth API to the hosts bluetooth API.

You'd need to resort to one of the projects mentioned.

closed 2023-10-27 by anthonws