technyon/nuki_hub #234: Problem configuring MQTT over TLS

magior, 2023-10-07:
I've been having problems trying to configure MQTT over TLS; I'm using nukihub with mosquitto MQTT server.

I'm using nukihub version 8.26 and configured it pasting PEM certificates of CA, client certificate and private key.

On mosquitto server, I receive this error:

1696663054: OpenSSL Error: error:14094412:SSL routines:ssl3_read_bytes:sslv3 alert bad certificate

On the serial port of nukihub I keep seeing:

[09:12:59]MQTT: Connecting with user: nukihub
[09:12:59]MQTT disconnected. Reason: TCP_DISCONNECTED
[09:12:59]MQTT connect failed, rc=Free Heap: 109916
[09:13:04]Attempting MQTT connection
[09:13:04]MQTT: Connecting with user: nukihub
[09:13:06]MQTT disconnected. Reason: TCP_DISCONNECTED
[09:13:06]MQTT connect failed, rc=Free Heap: 110440
[09:13:11]Attempting MQTT connection

I tried with the same certificate on a different client (MQTT explorer), and it connected without problems.

What am I doing wrong?

Marco

technyon, 2023-10-07:
What comes to mind is that your certificated exceed the maximum length. Did you verify the complete certificate is visible after copy/pasting it?

magior, 2023-10-07:
Thank you for your answer.
Unfortunately, yes, I've already checked, and they are complete.

mundschenk-at, 2023-10-07:
Is your CA root certificate perhaps using ECDSA? If so, that is the problem, it needs to be RSA. No idea why, but the ESP apparently does not like ECDSA root certificates (broker certificates work fine).

magior, 2023-10-10:
No, root CA built with easy-rsa and using RSA.

s00500, 2023-11-02:
Hm... I have the same issue here... getting this:

2023-11-02 17:35:17: New connection from 192.168.0.197:56301 on port 8883.
2023-11-02 17:35:17: Client <unknown> disconnected: Protocol error.

My current cert config is *

Any ideas ?

s00500, 2023-11-02:
I have only added the asterisk on top there. but another issue is that I have 2349 bytes of certificate...

technyon, 2023-11-04:
Any reasone why your certificate is that big? We tried to find a reasonable value that should cover most cases.

The maximum size could be adjusted, but it would take memory that could be used for other things.

s00500, 2023-11-05:
Ok, after looking into it I realised that just the actual certificate used on my broker is like that, the ca cert is smaller.

Still that does not solve the problem fully, I can post the ca.cert here if it helps

Or do any of you know how to get some more qualitative errors out of mosquitto to do debugging ?

technyon, 2023-12-27:
@s00500 Were you able to resolve your issue in the meantime? If not, what does "not solve the problem fully" mean?

s00500, 2023-12-28:
Hey @technyon 
I can still not get an ssl connection working. for now I am using an unencrypted connection (not very much prefered)

I still get errors in my mosquitto log (I am using the homeassistant addon)

For the next days I will be around a bit, returning around Jan 8, then I could do more testing

technyon, 2024-03-02:
@s00500 Were you able to investigate the issue further?

s00500, 2024-03-09:
Hm no still not... the esp is still in place but has unliked from the lock... I may have time to get back to it in 1,5w

technyon, 2024-03-09:
ok

s00500, 2024-03-19:
Ok, tried it again today, see my log here:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (551) esp_core_dump_flaKr��core dump partition found!
E (551) esp_core_dump_flash: No core dump partition found!
Nuki Hub version 8.33
IP address empty, falling back to DHCP.
IP configuration: DHCP
Hardware detect     : 1
Network device: Wifi only
MQTT over TLS.
-----BEGIN CERTIFICATE-----
MIIF8zCCA9ugAwIBAgIJAPzGhaodWEkmMA0GCSqGSIb3DQEBDQUAMFgxGTAXBgNV
BAMMEGhvbWVzZXJ2ZXIuYmFjaGkxDDAKBgNVBAoMA0xCUzEMMAoGA1UECwwDTEJT
MR8wHQYJKoZIhvcNAQkBFhBsdWthc0BsYnNmaWxtLmF0MB4XDTIyMDUxMDA3MTAy
N1oXDTMyMDUwNzA3MTAyN1owWDEZMBcGA1UEAwwQaG9tZXNlcnZlci5iYWNoaTEM
MAoGA1UECgwDTEJTMQwwCgYDVQQLDANMQlMxHzAdBgkqhkiG9w0BCQEWEGx1a2Fz
QGxic2ZpbG0uYXQwggIiMA0GCSqGSIb3DQEBAQUAA4ICDwAwggIKAoICAQDHYcqi
Is4/+T0mRWhR7tg1UI6qQaaaS5BzICEZHu3regRbMqPs+bsAvKH2wmX/INt+PuJY
F1l4mCF+xheGDx3nBjGjDpWOb4XMp+Eow03PT7nqvcf9n1GAGbU2jY9KpOunWuvg
m4oIINEg/Jv1TdnHMZlY47VaLYuoOV90d8TzPvMhqnoxmocKGXRPzddwOOcBGUqJ
vaFLsudHOTckI49W82myHFWKz4Qz+xIv80BNTOtT647VKDbMV7mGAFARPJtuWI7A
39Hw/2/yIwLAmvVtme6WJ66JFMQ4SMufvMktNopE+/6/LNQMwiM4rcbj6NKUQKAI
2Iybdz4IvgDF+lMledyv7Ur9dP9w+Y0aAdkiB1D1EUM0E59eErrv/dE5JpsWzbNA
clZTcZkhGu0Crt9n4j2xam16GNT3xw9KNj1E0RNX8KSb3QaMTbyPj+0Piy+ZyVMV
SlAw82zLHjBegZKhrh1LHyCRO4oHxVR/kH4okHHj1XPY5jO8Rc0/8aHA0HpkUkw8
sTBElj9RoHWK1XY+DR+x7n7thkqABS7SXRQJbbps9s8yP/RRajmoar9nPdxXWTmO
Qrt/vRGaM2/vxZFZhxZ2zQ0HCCX7pf/S8Ql/s7EkmH334aJc3c7rUlQbzt1ix+ZG
yj9KHJBhoA0I330NQ6VKh0eZVv8FuI/QRotpOQIDAQABo4G/MIG8MA8GA1UdEwEB
/wQFMAMBAf8wHQYDVR0OBBYEFG4Ad5C1kequschJOUTQRcEnNv+ZMIGJBgNVHSME
gYEwf4AUbgB3kLWR6q6xyEk5RNBFwSc2/5mhXKRaMFgxGTAXBgNVBAMMEGhvbWVz
ZXJ2ZXIuYmFjaGkxDDAKBgNVBAoMA0xCUzEMMAoGA1UECwwDTEJTMR8wHQYJKoZI
hvcNAQkBFhBsdWthc0BsYnNmaWxtLmF0ggkA/MaFqh1YSSYwDQYJKoZIhvcNAQEN
BQADggIBABPegr3n86ZpFCluK65e3t9l7RD3QV1xkLQaEMcDT90UwF7MXKMsuY7s
r3B+90gpClNUdXkbc1+Sos7LHedu7le9xGliXpl5SSjo+oEBPAhNiZcWLBMVXzS2
5oBGnnn/RKT/THYTztrDu8o93a3g9MsYC/lmOJWVCVVaj4CEX6bkH7GL9IPoakoQ
rbtstXY3NEYG4T1omthw7w31cfQokQ7gvdEOLOeERj+GyElbOTW/dkTlRO7Sa7+x
1N57cDzG1ifOHFq3/atYh7DxnmHBQttvadOh24YcVjq73OG07dm3l9PeWnSVwOpr
k4rM23UyDhj1IC+4gEBtfmwJ1+oHrOMLpR1UL+lD/4Fx2mU29JvRMhNACzaelqJM
zMUf5NMjPxjf6Ks4BtG0mBt1gN7S6zjQ03rgQwvkZg3UVrwa9isANuGsAxY+SLvt
I89YBZVu4iNxVz/VDS80TGOAxgDNU0WpNxxnEa+I1KOeGfQFQQMcuLI8BlLTxWrq
7b1d4ocI8HbGY/SWeaKX8lRQx4zBvt7SQCrmkafk25PviROPiXKPa6HIDOx4mz2t
EiwTLPiZa23ztIqhN0e9m53if9TTIh8ZKPNp3FtKcnsr7GjuGa1TfVqGqm75TvKT
lXW/aSuAAfybllGfHvoFVcGiibtraRoFpo5oPE14aPSmCFWuq2I2
-----END CERTIFICATE-----

*wm:AutoConnect 
*wm:Connecting to SAVED AP: BachiNetIoT
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: SUCCESS 
*wm:STA IP Address: 192.168.0.201

I am using the latest mosquitto addon on homeassistant-operationgsystem, and I see these errors there:

2024-03-19 23:18:55: New connection from 192.168.0.201:51895 on port 8883.
2024-03-19 23:18:55: OpenSSL Error[0]: error:14094412:SSL routines:ssl3_read_bytes:sslv3 alert bad certificate
2024-03-19 23:18:55: Client <unknown> disconnected: Protocol error.
2024-03-19 23:19:00: New connection from 192.168.0.201:51896 on port 8883.
2024-03-19 23:19:00: OpenSSL Error[0]: error:14094412:SSL routines:ssl3_read_bytes:sslv3 alert bad certificate
2024-03-19 23:19:00: Client <unknown> disconnected: Protocol error.

If I put in a * instead I get the following error:

2024-03-19 23:24:30: New connection from 192.168.0.201:64049 on port 8883.
2024-03-19 23:24:30: Client <unknown> disconnected: Protocol error.

Anything else I could try ?

s00500, 2024-03-19:
Note: I also upgraded to the latest nuki_hub sopftware using the webflasher

s00500, 2024-03-19:
Ok... so after a bit more testing I came to the conclusion that it is kinda save to switch out my current cert, as I have not a lot of devices using it. I generated a new ca and server key, then signed it like so:

```
# make a ca key
openssl genpkey -algorithm RSA -out ca.key

# make a CA cert
openssl req -new -x509 -days 3650 -key ca.key -out ca.crt -subj "/C=US/ST=YourState/L=YourCity/O=YourOrganization/OU=YourUnit/CN=YourCAName"

# make a server key
openssl genpkey -algorithm RSA -out server.key

# Make a sign request
openssl req -new -key server.key -out server.csr -subj "/C=US/ST=YourState/L=YourCity/O=YourOrganization/OU=YourUnit/CN=homeserver.bachi"

# sign it
 openssl x509 -req -in server.csr -CA ca.crt -CAkey ca.key -CAcreateserial -out server.crt -days 3650
```

(just make sure the CN matches!!!)

then I installed that in mosquitto and updated the certs, works fine now... I still dont understand exactly what the other issue was... But for me the issue is solved now.

iranl, 2024-05-19:
Added to README in #369

closed 2024-05-21 by technyon

jmsucasas, 2024-07-25:
@s00500 I have the same problem.  You have shown the way to generate the server certificates,  could you share how you have generate the client certificates to use in nukihub?

s00500, 2024-07-29:
@jmsucasas 
I dont use client certs in my case...

PR #369 (2024-05-21): Update README
Closes #234