technyon/nuki_hub #235: My Nuki HUB changes itself the firmware from 8.26 to the last installed 8.22

sblogoshs, 2023-10-11:
No matter how many times I have made the update from 8.22 to 8.26 without problems after a few days the 8.22 is active again and associated with this of course the error message in HomeAssistant.

sblogoshs, 2023-10-11:
The update is also displayed correctly in mqtt and I notice that whenever the error is displayed in the log of HomeAssistant. I know that this should not exist.

[image]
[image]

technyon, 2023-10-15:
Maybe your ESP is faulty. This firmware creates two partitions on the ESP, one for the old and one for the new firmware. In case the flash fails, it can fall back to the old firmware.

Maybe somehow your ESP can't boot the new firmware, so it falls back to the previous one. That would most likely mean some kind of hardware fault.

sblogoshs, 2023-10-15:
Thank you very much. That would of course be an explanation for it, let's see if it happens again with the new 8.27 and if so then I'll test another ESP.

technyon, 2023-12-27:
@sblogoshs Were you able to test another ESP?

sblogoshs, 2023-12-27:
Hi, since the installation of 8.27 the problem no longer exists. Thank you very much!