technyon/nuki_hub #236: ESP32/NUKIHub not reachable after some hours

KingSausage, 2023-10-12:
I flashed an ESP Wroom 32 and it occurs, that after a couple of hours, I cannot reach the nukihub and also don't see the esp in my lan.
After Reset it works fine and i can control the lock. Sometimes it is after 30min, sometimes after 2h.
I exchanged the ESP, the cable, the power socket and the position - no success. Because I have to hard-reset there is the standard message in the firmware-tab.

Any ideas? My next step would be to flash the ESP and setup the bridge without connecting the lock and look if this is a problem of the controller itself.

technyon, 2023-10-15:
Please post the content of the system information page (click firmware version).

KingSausage, 2023-10-15:
NUKI Hub version: 8.26
run: true
deviceId: 3704858090
deviceIdOp: 3704858090
mqttbroker: 192.XX.XX.XX
mqttport: 1886
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: 
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 
regAsApp: false
nrRetry: 
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.7.7
Lock hardware version: 11.1
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: Yes
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 1 minutes
Heap: 90492
Stack watermarks: nw: 6104, nuki: 760, pd: 268
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.

I always need to power-off/on the device.

technyon, 2023-10-15:
Try to

- enable "Restart on disconnect"
- Set "Restart if bluetooth beacons not received" to 60 seconds.

KingSausage, 2023-10-15:
Awesome, I will try this.
Give me a couple of days as I am not at home.

technyon, 2023-10-23:
It restarts because the connection is to the MQTT server is interrupted. So there's some network issue, what exactly it is I can't tell. Maybe unstable Wifi?

KingSausage, 2023-10-23:
I also changed to another board (ESP32 WROOM Mini D1) with the same result (via browser to FW 8.27). When I activated "Restart on Disconnect" it will do so almost every 2h, but stays alive. I also put the board next to the router, same result. Anyone has experience about the distance between the lock (BT) and the board, and the router and the board? Otherwise I will let him restart and catch the restart with a timer for about an minute before alarming (Telegram).

technyon, 2023-10-23:
It shouldn't reboot that often, however reboots are fast so you have a working setup I think. Is your router configured to automatically select the Wifi channel. Maybe try to set a fixed channel.

xaris33, 2023-11-17:
Mine did this because of a weak power supply, I changed it and it has been solid.

KingSausage, 2023-11-22:
Thank you for your advice!
I tried a different power source (a 65W) and unfortunately it also restart. I also changed my wifi to a fix channel. The only thing which changed (when using different power supply) is, that iobroker (mqtt server) log changed from saying "Message 105 delwted after xxx retries" to "Message 1 deleted after 11 retries".

KingSausage, 2023-11-23:
Unfortunately not, the ESP32 Mini D1does not have an Eth Port. I rather think about flashing a shelly 1 and put it behind the wall socket. But I hate the narrow place behind the socket :).

technyon, 2023-11-23:
The question is rather if you have LAN where you want to install the ESP. You'd have to buy an ESP with ethernet hardware of course, see readme.

xaris33, 2023-11-23:
Since you've tried everything else I'd try spinning up a different mqtt server if possible and see if the issue persists.

technyon, 2023-11-23:
I'd try setting a fixed Wifi channel on your 2.4 GHz network.

KingSausage, 2023-11-23:
I have already done before I changed the power supply :(.

technyon, 2023-11-24:
I can only repeat myself, using LAN makes a big difference. The ESPs Wifi stack isn't the most stable implementation, it is expected that it crashes every now and then, although every two hours sounds a bit too often. I'm running my ESP with W5500 hardware to connect to the LAN, and don't have any crashes whatsoever.

technyon, 2023-12-27:
@KingSausage Were you able to find a fix for your issue?

KingSausage, 2023-12-29:
@technyon unfortunately not. For my installation I don't want to/can't use LAN cable. I try to ignore the reboots :) and let my notify if timeout is longer than 2minutes.

technyon, 2023-12-29:
ok, but as far as I understand a t least it reboots and recovers?

KingSausage, 2023-12-29:
Yes, it "just" have a downtime of ca. 1:30min.

technyon, 2023-12-30:
That's long, the ESP should reboot within seconds. How exactly do you determine "down" ?

KingSausage, 2023-12-30:
Oh I think it can be a misunderstanding. I "watch" the downtime with mqtt online status via iobroker und let me notify (telegram) only when the "reboot" lasts longer than 1,5min. This happens 1-2 times per day, shorter "downtime" from the perspective of iobroker I write into the log. I would need to check it.

technyon, 2024-01-02:
Any chance you could get serial logs? It would be interesting to know why it takes that long ... the only thing I can think of is that it needs some time to figure out that it's disconnected. Just to confirm, "Restart on disconnect" is enabled in your current configuration?

KingSausage, 2024-01-02:
Yes, "restart on disconnecr" is enabled. I also think that the reboot itself does not last this long, it could rather be the "recognition".
Sure, I can connect the ESP to a PC and collect serial log, but you need ro give me some time for this setup.

Rubeast, 2024-02-03:
I'm having a similar issue.
I'm using a ESP32-WROOM-32D and my network devices are Unifi.

I have now enabled and set "Restart if bluetooth beacons not received" to 60 seconds
and enabled "Restart on disconnect".  

Will continue monitoring it.

technyon, 2024-02-04:
Hi,

I think it's to keep only one issue open so, don't worry about hijacking. The Wifi stack of the ESP32 is somewhat flaky. My recommendation is go to a LAN-based solution if possible. Espressif does fix issues though, which go into the Arduino core for ESP32 which I haven't updated in quite a while. In the binary below  I've updated from version 2.0.8 to version 2.0.14, please give it a try and let know if it improves your wifi connection.

[nuki_hub-8.32-pre-1.zip](https://github.com/technyon/nuki_hub/files/14155572/nuki_hub-8.32-pre-1.zip)

Rubeast, 2024-02-05:
I just installed 8.32-pre-1 and reconfigured the settings. I only configured my MQTT IP and paired the Nuki. The rest I kept default for now. Will keep you posted.

Rubeast, 2024-02-07:
I have to say it is much more stable now.
I'm monitoring it for 2 days now and I've only lost connection 1 time for about a minute.
[image]

KingSausage, 2024-02-07:
@Rubeast Awesome, i will try this as well as far as I can access my device.

Seinuh, 2024-02-11:
I had exactly the same problem... switched to olimex esp32-poe... now running another esp32 with the test firmware

technyon, 2024-02-12:
8.32 is now officially released

KingSausage, 2024-02-12:
> 8.32 is now officially released

Is it update-able via Update File or do we need to re-flash it?

Rubeast, 2024-02-13:
> Is it update-able via Update File or do we need to re-flash it?

You can upload the update file via the NukiHub interface. No need to re-flash it.

technyon, 2024-02-13:
Just upload via OTA

KingSausage, 2024-02-13:
I updated and will watch it for the next days. Thanks a lot for your effort!!!

technyon, 2024-03-02:
@Rubeast Are your problems fixed now?

Rubeast, 2024-03-03:
Looks like it. Thanks again!

technyon, 2024-03-04:
Great, I'll close this issue.

closed 2024-03-04 by technyon

Rubeast, 2024-03-04:
I'm not OP. But ok I guess 😛

jolichter, 2024-04-20:
Hi, even though the issue is closed, I would like to report a few things here.

**First of all,  I've been searching for a solution like this - thank you very much for your Nuki Hub!**

The Nuki Hub is running smoothly with my Nuki SL3P. I've disabled the WiFi and MQTT in the Nuki, because the Nuki calls home too frequently and consumes more power in WiFi mode.

Here's my experience with 4 different ESP32 devices:

My WiFi connection with an "ATOM Lite ESP32 IoT Development Kit" is linked with a signal strength of -55 dBm, which should be considered good enough. However, I notice that the connection has WiFi issues, maybe 2 times a day:

- Restart reason FW: RestartOnDisconnectWatchdog
- Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.

That's why I had also tried this with 3 other ESPs.

1. ESP32 NodeMCU ESP-WROOM-32 (USB-C)
2. NodeMCU ESP32 S (USB-Micro)
3. ESP32 Mini D1 NodeMCU (USB-C)

Although these have a larger wifi antenna on the board (at least visually), they showed a signal strength of -65 to -75 dBm and I still had more dropped connections. **So I decided to stick with the ATOM Lite ESP32.**

PS: The ESP32 devices are powered by a Pi-4 power supply and are located about 1.5 meters away from the Nuki SL3P. Of course I had also tried other 5V power supplies. Since some power supplies actually show a sawtooth on the line on the ozilooscope, I am currently using the Pi-4 power supply (5.1V, 3A), which shows a very clean DC voltage.

At the moment, the Nuki hub works best for me with an ATOM Lite ESP32, despite connection interruptions, which are apparently not critical?

System Information:

```
Nuki Hub version: 8.33
run: true
confVersion: 833
deviceId: 884844278
deviceIdOp: 2731148809
mqttbroker: 192.168.2.155
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
checkupdates: false
lockena: true
mqttpath: nuki
openerena: false
openercont: false
mqttoppath:
maxkpad:
opmaxkpad:
mqttca:
mqttcrt:
mqttkey:
hassdiscovery:
hassConfigUrl:
dhcpena: false
ipaddr: 192.168.2.42
ipsub: 255.255.255.0
ipgtw: 192.168.2.1
dnssrv:
nwhw: 1
nwwififb: false
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: true
rstbcn: 60
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0:
macb1:
macb2:
MQTT connected: Yes
Lock firmware version: 3.9.5
Lock hardware version: 5.12
Lock paired: Yes
Lock PIN set: No
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 443 minutes
Heap: 88608
Stack watermarks: nw: 6136, nuki: 568, pd: 232
Restart reason FW: RestartOnDisconnectWatchdog
Restart reason ESP: ESP_RST_SW: Software reset via esp_restart.
```

jolichter, 2024-04-22:
I also noticed this reset today:
```
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.
```
So the program had hung up and restarted?

iranl, 2024-04-23:
It is known that WIFI and BLE get disconnected sometimes. That is where the "Restart if bluetooth beacons not received", "Network Timeout until restart" and "Restart on disconnect" settings are for.

It is also known that the ESP sometimes crashes, the root cause is not exactly known (might be 1 cause, might by 10 different causes).

Because the ESP reboots in seconds and crashes/disconnects are infrequent it is not a major concern in most cases. 

Crashes are also hard to debug because the infrequent nature and that the ESP needs to be connected to serial and stack trace needs to be decoded to get usefull information.

jolichter, 2024-04-27:
Yes, of course. I have only tested my ESP32-S3 briefly, as soon as 8.34 is officially released I will test it longer with my Nuki. For the G0 button, I would have to solder a pin on this device, which was not necessary due to the Flash Tool.

Thanks again for your great work!

jolichter, 2024-04-27:
When version 8.34 is officially released, I will test it with a new [M5Stamp ESP32-S3](https://docs.m5stack.com/en/core/StampS3). I'm not sure if this will help because of the reboots, but it's worth a try.