technyon/nuki_hub #238: Mqtt Code not working

MrJamsn, 2023-10-17:
Hi, I am not sure if this is an issue of mqtt or nukihub. 
but I tried to set a code via the default_code field in Mqtt device setting. Though, code is not asked before execution. I also tried to change my nuki pin, no difference
any Idea? 
[image]
Thanks! Great Development!! Love it so far :)

MrJamsn, 2023-10-19:
> Please post the content of the system information page (click on the firmware version)

NUKI Hub version: 8.27
run: true
deviceId: 534405045
deviceIdOp: 534405045
mqttbroker: 192.168.178.41
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: true
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukibridge
nettmout: 60
restdisc: true
rstbcn: 60
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 0
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.6.9
Lock hardware version: 6.11
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 14 minutesHere we go :)____________

technyon, 2023-10-19:
MQTT is connected. I'm not sure what default code is about, but if you want to configure the PIN, set it in the NUKI Hub configuration portal.

alexdelprete, 2023-10-19:
> but I tried to set a code via the default_code field in Mqtt device setting. Though, code is not asked before execution.

where did you read that field is for setting a code that will be asked when calling lock/unlock/open? or it was simply an assumption? :)

MrJamsn, 2023-10-19:
I assumed based on this Article :)Lock Entity | Home Assistant Developer Docsdevelopers.home-assistant.io___

MrJamsn, 2023-10-19:
Okay. That‘s why I am asking :)Thought its software only, like how the mqtt lock identifies itselfs. Not a code wanted by the lock :)BR

"Some Locks"

alexdelprete, 2023-10-19:
> That‘s why I am asking :)

And that's why I'm answering. :)

closed 2023-12-27 by technyon