technyon/nuki_hub #242: Flashing with Web UI leaves bricked esp32

desheh, 2023-10-22:
Hey,

I tried flashing the latest nukihub on a brand new ESP32-DevKitC V4 from Aliexpress.

It finishes flashing (after long-pressing the BOOT button), but then after the reset is stuck in a loop showing this output every ~1s:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
```

Here's the output of the Chrome DevTools from the flashing process:

```
Try hard reset.
connect-221663ff.js?module:485 Chip type ESP32
connect-221663ff.js?module:485 Uploading stub...
connect-221663ff.js?module:485 Running stub...
connect-221663ff.js?module:485 Stub is now running...
connect-221663ff.js?module:485 Detecting Flash Size
connect-221663ff.js?module:485 FlashId: 0x16405E
connect-221663ff.js?module:485 Flash Manufacturer: 5e
connect-221663ff.js?module:485 Flash Device: 4016
connect-221663ff.js?module:485 Auto-detected Flash size: 4MB
connect-221663ff.js?module:485 Image header, Magic=0x01, FlashMode=0x00, FlashSizeFreq=0x00
connect-221663ff.js?module:485 Writing data with filesize: 8192. Compressed Size: 47
connect-221663ff.js?module:485 Took 11ms to write 47 bytes
connect-221663ff.js?module:485 Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
connect-221663ff.js?module:485 Image header, Magic=0xE9, FlashMode=0x02, FlashSizeFreq=0x2F
connect-221663ff.js?module:485 Writing data with filesize: 18560. Compressed Size: 12759
connect-221663ff.js?module:485 Took 1128ms to write 12759 bytes
connect-221663ff.js?module:485 Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
connect-221663ff.js?module:485 Image header, Magic=0xE9, FlashMode=0x02, FlashSizeFreq=0x2F
connect-221663ff.js?module:485 Writing data with filesize: 1659168. Compressed Size: 1047002
connect-221663ff.js?module:485 Took 92017ms to write 1047002 bytes
connect-221663ff.js?module:485 Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
connect-221663ff.js?module:485 Image header, Magic=0xAA, FlashMode=0x01, FlashSizeFreq=0x02
connect-221663ff.js?module:485 Writing data with filesize: 3072. Compressed Size: 129
connect-221663ff.js?module:485 Took 34ms to write 129 bytes
connect-221663ff.js?module:485 Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
connect-221663ff.js?module:485 DISCONNECT
connect-221663ff.js?module:485 HARD RESET
connect-221663ff.js?module:485 Try hard reset.
connect-221663ff.js?module:485 Initializing Improv Serial
```

I also tried flashing with the Espressif SDK CLI but it basically got stuck (I'm probably doing something wrong).

Any clue what's going on? Is the ESP32 I got not compatible (even though it shows up on the readme)?

technyon, 2023-10-23:
As far as I can see it's a regular ESP32 and it should be compatible. Did you try flashing a different firmware (e. g. tasmota) to exclude hardware problem?

desheh, 2023-10-29:
Tried flashing Tasmota/ESPHome and they both failed. I guess it's a busted esp32, thanks for the tip.

closed 2023-10-29 by desheh