technyon/nuki_hub #243: Template variable error: No first item, sequence was empty. when rendering '{{ (value_json|selectattr('type', 'eq', 'LockAction')|selectattr('action', 'in', ['Lock', 'Unlock', 'Unlatch'])|first).authorizationName }}'

DEVIDTR, 2023-10-30:
Please also add | default to this template setting as for Keypad Status. It also raises an issue:

Corrected: 

{"dev":{"ids":["nuki_30737e6c"],"mf":"Nuki","mdl":"SmartLock","name":"HomeLock"},"~":"nuki","name":"Keypad status","unique_id":"30737e6c_keypad_stats","stat_t":"~/lock/log","ent_cat":"diagnostic","ic":"mdi:drag-vertical","value_template":"{{ (value_json|selectattr('type', 'eq', 'KeypadAction')|first|default).completionStatus|default }}"}

technyon, 2023-10-31:
Check this binary please.

[nuki_hub-8.28-pre-1.zip](https://github.com/technyon/nuki_hub/files/13215922/nuki_hub-8.28-pre-1.zip)

technyon, 2023-11-04:
@DEVIDTR Does this fix the issue for you?

DEVIDTR, 2023-11-04:
Yes

You are receiving this email because you were mentioned.

closed 2024-01-02 by technyon