technyon/nuki_hub #245: Alexa / Google home support

sgouedj, 2023-11-06:
Hi,
is it possible to add a function to control the nuki hub with Alexa or Google home?
Thanks you

technyon, 2023-11-07:
The short answer is I'm not sure, I have no experience with both. But I guess it's nothing that's easy to add, it'll probably need software support on Alexa or google home.

sgouedj, 2023-11-07:
hello,
It's not very difficult, there is a dedicated library "fauxmoesp" for that.  I used it to order a shutter..but I don't know how to integrate it into your great nuki hub project. 
 Source and example here: https://randomnerdtutorials.com/alexa-echo-with-esp32-and-esp8266/
Thanks

mundschenk-at, 2023-11-07:
Personally I think that is something that should be done with a home automation solution, not on the ESP.

technyon, 2023-11-08:
I agree, it's not the philosophy behind NUKI Hub to support all kind of systems. You connect both NUKI Hub and Alexa/Google with your smart home solution (Home assistant, iobroker, openhab, ...) and whenever Alexa sends a command, your smart home solution will forward it to NUKI Hub via MQTT.

closed 2023-12-27 by technyon