technyon/nuki_hub #246: Cannot lock/unlock

giovi321, 2023-11-07:
I have installed the firmware successfully and it has been discovered by home-assistant.
However, when I try to lock/unlock from home-assistant (from the Developer tools page), it says that no entities are found for locking unlocking. However, the device is there in the mqtt integration. What am I doing wrong?

Arn0uDz, 2023-11-07:
Why are you doing this from the developer tool page instead of the devices page?

giovi321, 2023-11-07:
Because on the devices page there's no button to lock or unlock there

Arn0uDz, 2023-11-07:
Did you try to see what MQTT messages are being published? With MQTT explorer for example? If you don't have these options in the devices page then no need to even try the developer tools page.

technyon, 2023-11-07:
As a start, please publish the content of the system information page (click the firmware version).

giovi321, 2023-11-10:
Sorry for the delay, here is the information:
```
NUKI Hub version: 8.27
run: true
deviceId: 3276025239
deviceIdOp: 3276025239
mqttbroker: 192.168.1.65
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: true
ipaddr: 
ipsub: 
ipgtw: 
dnssrv: 
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: true
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.7.7
Lock hardware version: 6.11
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: Yes
Network device: Built-in Wifi
Uptime: 2 minutes
Heap: 89984
Stack watermarks: nw: 6104, nuki: 680, pd: 280
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_POWERON: Reset due to power-on event.
```

giovi321, 2023-11-10:
> Did you try to see what MQTT messages are being published? With MQTT explorer for example? If you don't have these options in the devices page then no need to even try the developer tools page.

I did, the messages seem fine but I don't see anything concerning actions

giovi321, 2023-11-10:
Here's what I see in the device configuration on home-assistant.
[image]

mundschenk-at, 2023-11-10:
@giovi321 You are missing the `lock` entity (I have no idea how that is even possible).

giovi321, 2023-11-10:
I have tried to remove and add back the mqtt integration, i have tried to unpair and re-pair the lock with the nuki hub but no luck. What else could I try?

mundschenk-at, 2023-11-10:
What version of Home Assistant are you running?

giovi321, 2023-11-10:
> What version of Home Assistant are you running?

Home Assistant 2023.7.3

mundschenk-at, 2023-11-10:
OK, you need to upgrade to at least HA 2023.8.0, as the `name` of the "main" entity is set to `None` in accordance with the new naming guidelines and that was not allowed before. (You will have some error regarding that MQTT discovery message in your HA log.)

giovi321, 2023-11-10:
Awesome! Thank you so much I will give it a try :)

closed 2023-11-11 by giovi321

giovi321, 2023-11-11:
Problem solved! Needed to upgrade :)