technyon/nuki_hub #247: Help controlling GPIO output in Home Assistant

asmaran85, 2023-11-11:
I connected a relay to pin P21 of the ESP32 on which NukiHub runs.
GPIO configuration -> GPIO 21 -> general output
In the configuration.yaml file I added the lines: 
`mqtt:
   switch:
    - unique_id: suoneria_citofono
      name: "Suoneria citofono"
      command_topic: "nuki/gpio/pin_21/state"
      payload_on: "1"
      payload_off: "0"`

In Home Assistant I see the new switch but I can't control it.
I'm not very good at mqtt. I tried with different payload (on/off, 1/0, high/low)

Thanks

technyon, 2023-11-11:
"nuki/gpio/pin_21/state" should be correct, unless you changed the MQTT path for the lock. Write "0" or "1" into this topic ... if that doesn't work, use an MQTT Client like MQTT explorer and monitor what happens on that topic if you toggle the switch.

asmaran85, 2023-11-11:
Is this good:
---------
mqtt:
  switch:
    - unique_id: suoneria_citofono
      name: "Suoneria citofono"
      command_topic: "nuki/gpio/pin_21/state"
      state_topic: "nuki/gpio/pin_21/state"
      payload_on: "1"
      payload_off: "0"
-----------

asmaran85, 2023-11-11:
it doesn'work

asmaran85, 2023-11-11:
Using the switch on home assistant the state changes right on mqtt explorer
[image]

But nothing happened on the relay (I moved it on pin23)

technyon, 2023-11-11:
It all looks correct. Could you get logs while switching the GPIO? You can use HTERM and connect the ESP to your PC, then look at the output while switching. Baud rate is 115200, you should line break to CR+LF.

asmaran85, 2023-11-11:
I changed the relay using a 2-relays-board and everything works!
Thank you very much!

technyon, 2023-11-12:
ok, glad it works now

closed 2023-11-12 by technyon