technyon/nuki_hub #254: NUKI Opener enabled even if deactivated

carloda, 2023-11-30:
Hi, I encountered a critical issue although I have not enabled "NUKI Opener enabled" on Home Assistant, I am given the possibility to execute the command via the icon highlighted in the screenshot, am I doing something wrong? Thank you

[image]
[image]

carloda, 2023-11-30:
> You know that the [Nuki Opener is a separate product](https://nuki.io/en/opener/)? Whar you see is the normal function of lock entities in HA.

@mundschenk-at I'm sorry, I was wrong, is it still inhibiting this function? Thanks and sorry again.

mundschenk-at, 2023-11-30:
@carloda Sorry, I don't understand what you mean. The feature in the config is for supporting the Nuki Opener and disabling it has therefore no effect on the functioning of the Nuki Smart Lock (which is a different product). There is a feature to make the lock "read-only", but I personally have not used that.