technyon/nuki_hub #255: Keypad-initiated unlock events not logged

noohi, 2023-12-01:
Hi,

is it expected that keypad unlocks are not tracked in the MQTT entries?
I tried to catch it, be it
nuki/maintenance/log
or
nuki/lock/action

But it is filled with useful unlatch entries only if using the Nuki itself directly, not if the unlock is initiated via the Keypad (fingerprint in this case).
Is that normal, to not see if the door were unlocked via the Keypad?

These are the events I tried, and they work if unlocking Nuki directly:
 When Nuki Lock Action changes to unlatch
 When Nuki Lock Log changes to Lock action received: unlatch
 When Nuki Lock Log changes to Querying lock state: unlocking
 When Nuki Lock Log changes to Querying lock state: unlatching
 When Nuki Lock Log changes to Querying lock state: Lock action received: unlatch

None of this happens if using the Keypad.

Any ideas?

Thank you.

Mikescotland, 2024-01-20:
It's i guess similar if you open it manually, sometimes it does register status, sometimes doesn't.

technyon, 2024-01-21:
It should always register, no matter if the status changed by keypad or manual operation.