technyon/nuki_hub #257: Feature request: measure keypad battery

giovi321, 2023-12-10:
Awesome work. Really.
I was wondering if there is any way to report the battery percentage of the keypad.

technyon, 2023-12-10:
Hi, no there's not. Regarding the keypad battery, the bluetooth API only reports one bit "Keypad battery critical", not the percentage.

giovi321, 2023-12-10:
Ooooh I see that's more than enough! However I cannot see it in home assistant with auto discovery, how con i add it?

technyon, 2023-12-27:
Hi. Sorry I lost track of this issue. If the keypad is paired to the lock, it should be visible via auto discovery. If not check what is published to MQTT when restarting the ESP. You'd need to use an MQTT client like MQTT explorer

iranl, 2024-04-23:
Can this be closed?
As mentioned by @technyon there is no percentage reported by the keypad in the API. The Nuki App also does not have this functionality.

So this can't be fixed.

If there is still a problem with auto discovery (usually a reboot of nuki hub or resaving config is enough to fix issues with auto discovery) please create a new issue.

giovi321, 2024-04-23:
Definitely let's close the topic. I wasn't aware of the possibility to read the state of "low battery", it's definitely enough.

closed 2024-04-23 by giovi321