technyon/nuki_hub #259: Compiling Error

ferdytao, 2023-12-21:
I'm trying to compile from source but I always got this error: 

```
Cmake Error at CMakeLists.txt:132 (target_link_arduino_libraries):
   Unknown CMake command "target_link_arduino_libraries".
```
I can't understand how to solve this error, tried to revert back to the original CMakeLists.txt but was the same error

technyon, 2023-12-22:
How exactly are you compiling? What's your build system, do you use the VM I provided?

In general you can follow these steps:

https://github.com/technyon/coffeetimer

Select minspiffs as the partitioning scheme.

ferdytao, 2023-12-22:
Thanks for your reply, I'm actually compiling for esp32c3 using your VM, I changed all the references to riscv compiler.
The previous error was resolved, now I got this error when launching ninja

This is an example:
```
[6/334] Building CXX object CMakeFiles/nuki_hub.dir/Network.cpp.o
FAILED: CMakeFiles/nuki_hub.dir/Network.cpp.o 
/home/nuki/.arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/esp-2021r2-patch5-8.4.0/bin/riscv32-esp-elf-g++  -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.6-dirty\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS   -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/newlib/platform_include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/freertos/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/freertos/include/esp_additions/freertos -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/freertos/port/riscv/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/freertos/include/esp_additions -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_hw_support/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_hw_support/include/soc -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_hw_support/include/soc/esp32c3 -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3 -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3/private_include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/heap/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/log/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/lwip/include/apps -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/lwip/include/apps/sntp -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/lwip/lwip/src/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/lwip/port/esp32/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/lwip/port/esp32/include/arch -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/soc/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/soc/esp32c3 -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/soc/esp32c3/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/hal/esp32c3/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/hal/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/hal/platform_port/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_rom/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_rom/include/esp32c3 -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_rom/esp32c3 -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_common/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_system/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_system/port/soc -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_system/port/include/riscv -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_system/port/public_compat -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/riscv/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/driver/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/driver/esp32c3/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_pm/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_ringbuf/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/efuse/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/efuse/esp32c3/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/vfs/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_wifi/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_event/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_netif/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_eth/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/tcpip_adapter/include -I/home/nuki/.arduino15/packages/esp32/hardware/esp32/2.0.14/tools/sdk/esp32c3/include/esp_phy [... cut ...]
     const String redact(const uint i) const
                  ^~~~~~
/home/nuki/projects/nuki_hub/PreferencesKeys.h:110:18: note: candidate: 'const String DebugPreferences::redact(int64_t) const'
     const String redact(const int64_t i) const
                  ^~~~~~
/home/nuki/projects/nuki_hub/PreferencesKeys.h:114:18: note: candidate: 'const String DebugPreferences::redact(uint64_t) const'
     const String redact(const uint64_t i) const
                  ^~~~~~
ninja: build stopped: subcommand failed.
```

technyon, 2023-12-23:
It seems the risc-v compiler can's manage the method overloads. Taking a guess, maybe try to make them more specific (see code below).

Why are you using a different compiler in the first place, any specific requirements you have?

Another thing, the c3 is a single core chip. You'll have to rebuild your own "Arduino core for ESP32" built for one CPU core only, otherwise your binary will crash.

--

```
    const String redact(const String s) const
    {
        return s == "" ? "" : "***";
    }
    const String redact(const int32_t i) const
    {
        return i == 0 ? "" : "***";
    }
    const String redact(const uint32_t i) const
    {
        return i == 0 ? "" : "***";
    }
    const String redact(const int64_t i) const
    {
        return i == 0 ? "" : "***";
    }
    const String redact(const uint64_t i) const
    {
        return i == 0 ? "" : "***";
    }
```

ferdytao, 2023-12-23:
Ok thanks for the tips, I'm not using a different compiler, just following the instructions on Github `cmake .. -G Ninja` and `ninja` to compile it.

I will try to find the instruction how to rebuild the Arduino core for my esp32c3 board. I'm not a programmer so I'm studying step by step all things are going to happen.

PR #341 (2024-04-25): Add C3, S3 and Solo1 support + Build using PlatformIO
-ESP32-S3 build was flashed and tested with success.
-ESP32-C3 and ESP32-Solo1 are untested, but build ok.

Credits for PlatformIO support go to @lucaoliano.
Some changes where required because the bootloader address of the ESP32-S3 and C3 differs from the ESP32.

Closes #199 
Closes #201 
Closes #259 

Leaves CMake support intact.
Artifacts from Github Actions can be used for release zip files and merged binaries in the ZIP files can be used for Webflash.
Could benefit from a release workflow to automate adding binaries to releases/webflas [... cut]