technyon/nuki_hub #261: configuration/autoLock not showing up in Home Assistant

bjiirn, 2023-12-23:
In Home Assistant the AutoLock Function is not showing up in the Configurations.

Does anyone have an idea why that could be or how to fix that?

When i Look at the MQTT messages it shows up.
I don't know much about MQTT and how Auto Discovery works tho.

[image]
[image]

technyon, 2023-12-24:
I think auto discovery for auto lock is not implemented yet, it could be added though.

bjiirn, 2023-12-24:
Ah, ok. Thanks for the clarification. 
It would be grate if that were part of the DeviceConfig,
but for now i got it working as a separate entity with this in my configuration.yaml :
```
mqtt:
  - switch:
      unique_id: nuki_autolock
      name: "Nuki Auto Lock"
      state_topic: "nuki/configuration/autoLock"
      command_topic: "nuki/configuration/autoLock"
      payload_on: "1"
      payload_off: "0"
      state_on: "1"
      state_off: "0"
```

closed 2023-12-24 by bjiirn

mundschenk-at, 2023-12-24:
You can have the switch appear in your device by adding a `device` section to your YAML:

```yaml
  device:
    identifiers: nuki_<device>
    name: <Device Name>
```