technyon/nuki_hub #262: Lots of lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2) in the logs

Appoxus, 2023-12-27:
When trying to lock/unlock I get a lot of these messages in the log, resulting in a timeout and the lock sometimes doesn't react.

```
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Lock: Maximum number of retries exceeded, aborting.
```

technyon, 2023-12-29:
It looks like a problem with reception. How close is your ESP to the lock?

technyon, 2024-01-04:
@Appoxus Any update on this?

Appoxus, 2024-01-04:
Sorry, forgot to answer.
It was in plain sight. Approx. 2-3 Meters away.

technyon, 2024-01-04:
Strange ... that's not much, I can only guess:

- Just for testing, can you move it closer?
- Do you have another ESP to test? Maybe it's a hardware problem
- What kind of power supply do you use? Maybe the ESP gets some noise from there.
- Is the firmware of your lock up-to-date?

Appoxus, 2024-01-04:
- I have to reflash to ESP again, can take a while.
- Nope, it was my last spare one.
- I used my Laptop as power source with a long USB cable. I think that could be the cause.
- Yep, Firmware is up to date.

technyon, 2024-01-05:
One more thing that sometimes helps around quirks with NUKI devices is to just factory reset them. Quite annoying I know.

Appoxus, 2024-01-05:
I flashed the ESP again and used a separate Power supply.
Since I didnt use my Notebook for Power supply, I couldnt get the logs to see if the error happened again.

Nontheless my intention was to test if nuki hub ist faster than the nuki bridge. I couldnt tell if theres a difference, at least in my environment.

I'll close the issue since I wont do further testing.
Thank you for your support.