technyon/nuki_hub #263: False positives in reported critical battery states

bcutter, 2023-12-27:
Strange behavior, seen this a few times (every few weeks roughly) already, today I took the time to have a look at it:

`binary_sensor.lock_door_a_battery_low` became active, so did the `binary_sensor.lock_door_a_keypad_battery_low` entity:

[image]

[image]

Please note this is the lock battery entity (the keypad has none according to https://github.com/technyon/nuki_hub/issues/257) which is obviously NOT critical:

[image]

I also checked the lock and Keypad battery state with the Nuki iOS app immediately (roughly 16:53):
- lock: exactly as provided by Nuki-Hub to HA (as shown above)
- keypad: green text "ausreichend geladen" (in English basically: battery is okay)

I'm wondering
1. where this comes from
2. how I could debug this once it happens again (where exactly to look at in MQTT Explorer at the time this happens again?)

Of course it can be the lock and or Keypad (at the same time? hmm...) itself, but as I checked the iOS app immediately and everything was looking fine there...

My theory is: lock and/or keypad were sending rubbish for a very short time. NH of course catched it. Exactly one hour later the correct state has been read by NH from the lock and transfered via MQTT to HA again. Does that make sense and match these NH settings?

```
NUKI Hub version: 8.26
...
lockStInterval: 3600
configInterval: 3600
batInterval: 7200
kpInterval: 1800
kpEnabled: true
...
```

technyon, 2023-12-29:
Hi,

The battery percentage/critical information is actually sent in the lock state request (lockStInterval) ... so it should correct itself either after the timeout of one hour, or if you execute some lock action.

Why this would be incorrectly reported I can only guess. Bluetooth should have enough checks to prevent corrrupt data to be received. So either the lock sends the wrong data sometimes, or it gets corrupted after receiving it .... both very hard for me to reproduce.

Other than this everything works fine?

bcutter, 2023-12-29:
> Other than this everything works fine?

Yes. Very stable, no issues.

What about debug options to check if it's actually sent from the lock itself?

technyon, 2023-12-30:
This firmware will the relevant fields to serial and/or mqtt. To decode, check the API documentation:

https://developer.nuki.io/page/nuki-smart-lock-api-2/2#heading--keyturner-states

Fields "Critical Battery state" and "Accessory Battery State".

[nuki_hub-8.29-dbg-1.zip](https://github.com/technyon/nuki_hub/files/13797087/nuki_hub-8.29-dbg-1.zip)

bcutter, 2023-12-30:
I need to update my HA Core first (still running NH 8.26 because of #227) to not run into https://github.com/technyon/nuki_hub/issues/227 again.

cagnulein, 2024-02-06:
[image]
is this could be about this? the battery dropped to 60% and then up again to more than 90%
Never saw a similar behaviour in the last 3 weeks

technyon, 2024-02-06:
Unfortunately I have no good explanation. NUKI Hub just forwards to MQTT whatever battery level/state the lock sends. It  would be interesting to know what level the App would report at the same time, but that's gound to be quite hard to catch.

cagnulein, 2024-02-06:
@technyon Makes sense. I will try to catch this information next time it happens. I was scared because I thought the battery wouldn't last a month! :D

technyon, 2024-03-02:
@cagnulein Were you able to investigate the issue further?

cagnulein, 2024-03-02:
@technyon didn't have issue again after that

technyon, 2024-03-02:
ok, I'll close for now then. Should the issue come up again, let me know.

closed 2024-03-02 by technyon

bcutter, 2024-03-11:
I know this issue is closed. And I did not see this issue for a quite long time. But just now two things happened:
1) "battery low" entity got triggered
[image]

2) Very very strangely: the "battery level" value is, well... a "bit" strange - I mean 100 % is great, and more is even better right 🤣
[image]

At the same time the "battery voltage" is completely fine:
[image]

I checked MQTT:
[image]

At the same time directly in Nuki iOS app:
[image]

Seems like this wrong reading happened right during/after a NH restart:
[image]

Another (manual) NH restart "solved" the current incident:
[image]

[image]

---

I lost a bit track of this issue, not sure if there was an update meanwhile trying to handle those obviously wrong values (admitted: the battery level > 100 % thing is new, could be filtered...). Disclosure: I'm on NH v8.26.
--> Edit: ah I see, it seems like 8.27 had some juice in there to handle this, right? Wanted to share the > 100 % experience anyway.

technyon, 2024-03-11:
If a level above 100% is reported, this must be a fluke in the Nuki firmware. There are 6 bits representing the battery level, you multiply the value by 2 and that's the level. Sure it could be filtered.