technyon/nuki_hub #264: Nuki Opener: lock/state changing between ring and locked in seconds

MarcAroni81, 2023-12-28:
Hello,

I have paired the Nuki Opner to the Nuki Hub v 8.28 (mqtt nuki opener path set to: "nukiopener")
Now I receive in seconds intervall the mqtt "nukiopener/lock/state" = "locked" and "ring" which ends up flickering the "ring detected" MQTT sensor. 

Any ideas where to start debugging this behavior? 

What I already did: 
- flashed the ESP32 device and re-installed nuki_hub
- restarted HA as well as nuki_hub several times
- removed the MQTT device and waited for it to be discovered again

Many thanks for your help!
Marcus

MarcAroni81, 2023-12-28:
I fixed it myself:

I disconnected the Nuki Opener from the Hub,
resetted Nuki Opener Device (Factory Reset) and did the initial setup 
re-paired the device 

= 
fixed

closed 2023-12-28 by MarcAroni81

technyon, 2023-12-29:
Glad you could fix it. I had a similar issue some time ago that could only be fixed by a factory reset.