technyon/nuki_hub #265: No beacons from HA app in MQTT presence

xaris33, 2024-01-04:
Hello,

As per title, I'm unable to pickup the ibeacon from the HA app in MQTT to use for automations. That would be great as the nuki sometimes take too long to auto open the door. 

At the moment I'm running ESP presence on another ESP32 and it's much faster, I'd like to just use one if possible.

I just tried 8.29 as the changelog mentions the ibeacons but no change.

Sample mqtt messages below (I have 2 android phones, both not showing):

18:d9:8f:56:95:e7;HONOR Band 5-5E7;-79
54:d2:72:20:58:dd;Nuki_312058DD;-62
7e:e9:87:f1:41:4c;Galaxy Watch4 (6DCD);-98
d8:88:cd:22:b6:b7;Mi Smart Band 4;-83
d8:e0:e1:2a:82:75;[TV]Bedroom;-78

Thank you!

xaris33, 2024-01-06:
This worked, thank you. Although I have the latest version of the app the minor id was 1.

Unfortunately I have no idea how to make automations out of this long string of devices in mqtt, is there a tutorial available for HA?

technyon, 2024-01-06:
Not that I know off. @mundschenk-at @alexdelprete Maybe you can help better from this point. 

Other than that maybe ask in the HA forum.

mundschenk-at, 2024-01-06:
I have not experimented further than https://github.com/technyon/nuki_hub/issues/159#issuecomment-1554358294 with presence detection as I am using iOS and the app is more limited there for beacons (and connection with Apple devices in general is much easier to do in [ESPresense](https://espresense.com)). Since this is mostly about HA templating, I'd also refer the question to the forum.

alexdelprete, 2024-01-06:
> @mundschenk-at @alexdelprete Maybe you can help better from this point.

Sorry, I don't use bluetooth presence detection, it's not reliable, so I can't help much.

In any case, I wouldn't have implemented any of this because there will be a LOT of issues opened because of that not working (like in other BT presence detection projects), and I think Nuki Lock should only focus on its main purpose: managing the Nuki Lock.

Anyway, it seems strange that BT beacons of two mobiles are not being picked up, while other devices are. And that is why I never liked this approach for presence detection: not reliable.

xaris33, 2024-01-06:
Thank you all for your replies.

@alexdelprete the detection issue was resolved easily, it was just the ble transmitter being configured incorrectly as above.

My only complaint with the lock is the auto unlock speed, most of the time you have to wait outside the door for it to open for several seconds. Not very star trek :) This is probably due to energy savings for the battery.

I have set up a 2nd esp32 with espresence and it has been great, the door opens just as you walk up to it most of the time and really enhances the experience. For this I feel that adding the feature would interest many users. 

My problem at the moment is that the data in the presence topic is in CSV form, when I think we would need some sort of JSON like espresence. I have no idea how to match the phone's UUID in the payload so I can use it in an automation.

technyon, 2024-01-07:
Hi,

JSON is quite a bloated format, and putting the ESPs capabilities are limited. This is already a problem when publishing the log entries, when I go higher than five, it won't work.

You should keep in mind that beacons are transmitted without any encryption. Anyone close to you can receive them, copy them, and send those same beacons to unlock your door. If that's a realistic scenario is another question, I think in almost all cases they would use force to enter your home if they want to. Just mentioning it so you are aware.

alexdelprete, 2024-01-07:
> My only complaint with the lock is the auto unlock speed, most of the time you have to wait outside the door for it to open for several seconds. Not very star trek :)

There are better ways to automate the unlock, but remember that the house door (at least for me) is a highly-secure scenario, so I don't trust anything that might have the remote possibility of being hacked. BLE beacons are not encrypted, as Jan said, so I would NEVER rely on something like that.

ESPresence is faster because it ONLY DOES one thing. The Nuki Hub does WiFi+BT+MQTT+Ethernet, and ESP developers wrote in several posts that the ESP32 barely manages to do it, it requires a lot of optimizations and avoiding expensive functions or memory-hungry routines.

It took quite a while to reach a reliable level of the firmware, Jan did a great job and we tested for months until it reached a good maturity level.

If I was Jan, I'd never have implemented any presence functionality, for me it's totally outside the scope of the project, while there are specialized projects for that use-case.

technyon, 2024-01-07:
Well you know, as it often is the case I implemented it because I'm using it myself.

I didn't except it to be be that complicated for CSV to be used, considering it's one one of the most simple formats there is ... but the home assistant devs decided to base all the input on json. In iobroker I wrote a simple script that parses the input line by line, then does a string split ";", no big deal.

I think as this is not main feature of the firmware and the limitations mentioned, I won't change this. It's easy enough to run a secondary ESP with a firmware that's more compatible with HA if need be.

mundschenk-at, 2024-01-07:
It's not that hard in HA either, the comment I linked shows one way to do it. (But there is more tooling for JSON in HA, that is correct.)

xaris33, 2024-01-07:
All valid points, thank you.

For anyone else interested in such an approach, my automation is:

Wait for HA person tracker to change to home.

Actions:
1. Turn on ble transmitter of relevant phone.
2. Wait for ble to be detected (either use espresence for this or Nukihub)
3. Open door 
4. Turn off ble transmitter.

This maximises battery life on the phone and only allows a small window for the door to be opened for security purposes.

If you have multiple users then make an automation for each but add actions so that the first automation to run successfully stops the automations of the other users for a minute or 2 to  prevent multiple opens when arriving together.

There may be a more elegant way to do this or within a single automation :)

closed 2024-01-07 by xaris33