technyon/nuki_hub #266: presence detection

pilot1981, 2024-01-05:
hi,
how works presence detection?

mundschenk-at, 2024-01-06:
That's not really an issue with NukiHub, and it's also already discussed in multiple places #159, #265.

pilot1981, 2024-01-06:
Sorry, I see new feature in lastest firmware version, but in issue linked I see opened status…I don’t understand if this feature is already available…

mundschenk-at, 2024-01-06:
It's not a new feature, just a tweak on an existing one. And there is such a thing as ”search” on GitHub, and a [Home Assistant specific topic in the HA forums](https://community.home-assistant.io/t/nuki-hub-firmware-for-the-esp32/417572) (which I'm assuming you want the presence detection to work with?).

iranl, 2024-05-17:
Presence detection is covered in the README (https://github.com/technyon/nuki_hub?tab=readme-ov-file#misc). How to use these values is discussed extensively on the Home Assistant forum and previous issues to which @mundschenk-at has linked.

closed 2024-05-17 by iranl