technyon/nuki_hub #267: ATOM Lite ESP32 needs to rebot after a while but do not crash

robitsrl, 2024-01-07:
I have Nuki Hub on ATOM Lite ESP32 and I notice that sometime (when i need it) do not open the door.
I have to reboot the hub from HA.
to fix this i made and autoreboot every few hours but still when i need it does not works.
I'm using version 8.2.8 with a smart lock 3 pro setup with both hub and built in wi-fi
I see there is a new version 8.2.9 with presence detection. How does it works ?
i use both hub and wifi build it because i like the auto unlock feature of the app.
Does the 8.2.9 presence detection works like the auto-unlock ?

technyon, 2024-01-07:
Regarding presence detection, please check issue #266. It's not a new feature.

Please post the content of the system information page (click on the firmware version)

nurimoca, 2024-01-15:
> I have Nuki Hub on ATOM Lite ESP32 and I notice that sometime (when i need it) do not open the door. I have to reboot the hub from HA. to fix this i made and autoreboot every few hours but still [...]

Same with me. Any solution?I created a automation,  as a walk-around that resets the hub every hour, and it seems to work.

technyon, 2024-01-16:
I need some more information, like what kind of automatic reboots are configured. The easiest way to provide this is to post the system information page.

eikaramba, 2024-01-23:
I just read this (german) article, which shows how the esp32 is affected by a bad power voltage and can lead to reboots.
https://arduino-hannover.de/2018/07/25/die-tuecken-der-esp32-stromversorgung/ (use google for translation)

TLDR: use a high quality power supply(hard to measure without a oziloscope), use short usb cable with a big diameter, solder a 47-µF-Tantalcondensator to the esp32

iranl, 2024-05-26:
Closing as this issue has been stale for 4 months and reported on an out of date Nuki Hub version. 
Furthermore using Nuki Hub and Wifi together is only possible if you pair NukiHub as an app instead of a bridge and is strongly discouraged as it is not very stable and drains the battery of the Nuki.
Please reopen if this issue still occurs on 8.34 when paired as a bridge and post the contents of the info page as suggested by @technyon

closed 2024-05-26 by iranl