technyon/nuki_hub #269: feature request: OTA Updates & Update info

miggi92, 2024-01-09:
It would be nice to have a OTA Update Button and an Information if there is a new update available.

Example:
[image]

Update Button:
[image]

technyon, 2024-01-09:
See issue #125

miggi92, 2024-01-10:
> See issue #125

ah thanks. Sry didn't see that issue.

iranl, 2024-02-09:
#293 should partially fix this as it will enable automatic update checks that show the latest release in the web interface, MQTT and Home Assistant.

I don't believe true (semi)automatic updating that will download and install the latest binary is feasable at this time.

Propose merging #293 will close this for the time being.

iranl, 2024-02-18:
@technyon @miggi92 Can this be closed with the modifications in the latest release?

miggi92, 2024-02-23:
@iranl  yes. I'll close it.

closed 2024-02-23 by miggi92