technyon/nuki_hub #270: MQTT topic content for nuki/lock/log is cut-off

micronax, 2024-01-14:
Hi,
thanks for building this super nice piece of code. The integration works like a charm.

I just encountered a problem with the log supplied within the MQTT topic `nuki/lock/log`:

[image]

The JSON is somewhere just cut-off with some strange byte characters which makes it impossible to parse.

Is this kind of an overlow?

technyon, 2024-01-15:
It's not exactly an overflow. There's a buffer allocated of 4096 bytes. The JSON serializer checks against this maximum size, and in your case has hit the limit. There's a flag in the serializer to check for that, in that case the last log entry should be removed ... or the easy fix maybe is to increase the buffer size. I'd need to to some math what the maximum size could be, considering you're using the maximum length for the keypad entry names.

micronax, 2024-01-15:
Thank you 🙏 

> considering you're using the maximum length for the keypad entry names

Is there anything I can change on my side to make it work out-of-the-box?

I only have one keypad permission right now and it's just called "Zutrittscode".

technyon, 2024-01-16:
You'll find other things than keypad entries in the log. Any time you a lock action is executed, it appears in the log. Did you assign any long names (Android/IOS App, Fobs)? Also could you copy/paste the payload into a text editor and see if it really is longer than 4095 characterrs (4096 minus one for the termination character).

micronax, 2024-01-16:
Nope, my setup is pretty new. Just this one access code, no fobs, no other users etc.

micronax, 2024-01-17:
I use the Mosquitto addon for Home Assistant. I also encountered that some lock-related topics are not updating for hours while eg. presence is updated every few seconds, also after changing eg. the lock state interval to 300 seconds etc.

Do you need some config insights?

technyon, 2024-01-18:
Let's check first what's going on on ESP side. Can you get serial logs? This firmware prints the log entry count when publishing, it should be no more than 5.

[nuki_hub-8.29-dbg-1.zip](https://github.com/technyon/nuki_hub/files/13977133/nuki_hub-8.29-dbg-1.zip)

micronax, 2024-01-25:
Somehow I wasn't able to flash that binary to my esp32 yet. I will try again tonight.

More insights:
- I had to exchange my Nuki lock with a new one after I had issues with the lock mechanism. So now I have another and brand-new Nuki 4 (non-pro) and the same problem persists with the cut-off logs
- I re-installed the esp32-nukihub like 3 times with the default binary from the installer, same problem persists

Maybe its a general problem with the Nuki 4 locks?

technyon, 2024-01-25:
Very strange. Getting those serial logs would be helpful. If it's a Nuki 4 issue, it's a bug in the Nuki 4 firmware.

micronax, 2024-01-25:
Here comes the serial logs:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (528) esp_core_dump_flaKr��core dump partition found!
E (528) esp_core_dump_flash: No core dump partition found!
NUKI Hub version 8.29-dbg-1
IP address empty, falling back to DHCP.
IP configuration: DHCP
Hardware detect     : 1
Network device: Wifi only
MQTT without TLS.
*wm:AutoConnect
*wm:Connecting to SAVED AP: <redacted>
*wm:connectTimeout not set, ESP waitForConnectResult...
*wm:AutoConnect: SUCCESS
*wm:STA IP Address: <redacted>
WiFi connected: <redacted>
Host name: nukihub
MQTT Broker: <redacted>
NUKI Lock enabled
Device id lock: <redacted>
Lock state interval: 300 | Battery interval: 1800 | Publish auth data: yes
NUKI Opener disabled
Presence detection timeout (ms): 60000
Attempting MQTT connection
MQTT: Connecting with user: <redacted>
MQTT connected
Nuki lock start pairing
Nuki paired
Querying lock state: unlocked
Log entry count: 22
Querying lock battery state: success
Reading config. Result: success
Reading advanced config. Result: success
HASS setup for lock completed.
HASS setup for lock completed.
```

jmoransalama, 2024-01-25:
I'm having the same issue with my Nuki4 Pro lock.

NUKI Hub info:
```
NUKI Hub version: 8.29
run: true
deviceId: <REDACTED>
deviceIdOp: <REDACTED>
mqttbroker: 192.168.8.10
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 6
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: false
ipaddr: 192.168.8.56
ipsub: 255.255.255.0
ipgtw: 192.168.8.1
dnssrv: 1.0.0.1
nwhw: 1
rssipb: 60
hostname: nukihub
nettmout: -1
restdisc: false
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: true
accLvl: 0
regAsApp: false
nrRetry: 3
rtryDelay: 100
crdusr: ***
crdpass: ***
pubauth: true
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 4.0.37
Lock hardware version: 7.0
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: Yes
Lock has keypad: Yes
Network device: Built-in Wifi
Uptime: 913 minutes
Heap: 65824
Stack watermarks: nw: 6104, nuki: 600, pd: 232
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_INT_WDT: Reset (software or hardware) due to interrupt watchdog
```

MQTT value of /nuki/lock/log:
```
[{"index":565,"authorizationId":209924,"authorizationName":"NukiHub","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":0,"timeMinute":40,"timeSecond":55,"type":"LockAction","action":"Lock","trigger":"system","completionStatus":"success"},{"index":566,"authorizationId":18551808,"authorizationName":"","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":10,"timeMinute":47,"timeSecond":58,"type":"LockAction","action":"LockNgo","trigger":"button","completionStatus":"success"},{"index":567,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":10,"timeMinute":47,"timeSecond":59,"type":"DoorSensor","action":"DoorOpened","completionStatus":"undefined"},{"index":568,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":10,"timeMinute":48,"timeSecond":6,"type":"DoorSensor","action":"DoorClosed","completionStatus":"undefined"},{"index":569,"authorizationId":133518,"authorizationName":"Nuki Keypad","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":14,"timeMinute":41,"timeSecond":9,"type":"KeypadAction","action":"Unknown","completionStatus":"undefined"},{"index":570,"authorizationId":133518,"authorizationName":"Nuki Keypad","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":14,"timeMinute":41,"timeSecond":15,"type":"KeypadAction","action":"Unknown","completionStatus":"undefined"},{"index":571,"authorizationId":133518,"authorizationName":"Joaco","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":14,"timeMinute":41,"timeSecond":39,"type":"KeypadAction","action":"Unlatch","completionStatus":"success"},{"index":572,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":14,"timeMinute":41,"timeSecond":39,"type":"DoorSensor","action":"DoorOpened","completionStatus":"undefined"},{"index":573,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":14,"timeMinute":42,"timeSecond":2,"type":"DoorSensor","action":"DoorClosed","completionStatus":"undefined"},{"index":574,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":15,"timeMinute":31,"timeSecond":3,"type":"DoorSensor","action":"DoorOpened","completionStatus":"undefined"},{"index":575,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":15,"timeMinute":31,"timeSecond":5,"type":"DoorSensor","action":"DoorClosed","completionStatus":"undefined"},{"index":576,"authorizationId":133518,"authorizationName":"Joaco","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":15,"timeMinute":42,"timeSecond":30,"type":"KeypadAction","action":"Unlatch","completionStatus":"success"},{"index":577,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":15,"timeMinute":42,"timeSecond":30,"type":"DoorSensor","action":"DoorOpened","completionStatus":"undefined"},{"index":578,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":15,"timeMinute":42,"timeSecond":30,"type":"DoorSensor","action":"DoorClosed","completionStatus":"undefined"},{"index":579,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":15,"timeMinute":57,"timeSecond":52,"type":"DoorSensor","action":"DoorOpened","completionStatus":"undefined"},{"index":580,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":15,"timeMinute":58,"timeSecond":1,"type":"DoorSensor","action":"DoorClosed","completionStatus":"undefined"},{"index":581,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":17,"timeMinute":37,"timeSecond":17,"type":"DoorSensor","action":"DoorOpened","completionStatus":"undefined"},{"index":582,"authorizationId":86077,"authorizationName":"2131887821","timeYear":2024,"timeMonth":1,"timeDay":13,"timeHour":17,"timeMinutxV�� 
[... 13 lines cut ...]
[... cut ...]
2024-01-25 18:52:40.218 WARNING (MainThread) [homeassistant.components.mqtt.discovery] Unable to parse JSON smartlock: '{"dev":{"ids":["nuki_38354535"],"mf":"Nuki","mdl":"SmartLock","name":"E50local"},"~":"nuki","name":null,"unique_id":"38354535_lock","cmd_t":"~/lock/action","avty":{"t":"~/maintenance/mqttConnectionState"},"pl_lock":"lock","pl_unlk":"unlock","pl_open":"unlatch","stat_t":"~/lock/binaryState","stat_locked":"locked","stat_unlocked":"unlocked","opt":"false"}'
```

If I delete the autodiscovery topics on the MQTT broker, the warnings on home-assistant go away for a few hours, but then reappear after some time.

I only have one NUKI Lock and one NUKI Hub.

technyon, 2024-01-26:
This is your problem:

`Log entry count: 22`

Only 5 entries are requested from the lock, so there should be a maximum of 5 entries. I suspect a bug in the firmware, especially if it 's only reported for 4.0 locks.

Let me get in touch with the devs of the library that controls the lock. Problem is though we could ask in the NUKI developer forum, but lately there seems to be noone answering there.

micronax, 2024-01-26:
So the JSON is already delivered cut-off from that other library?

Which library is that?

technyon, 2024-01-26:
Yes that's my fork, altough there should be no difference to the main project at this time, except maybe for a few missing commits.

That library doesn't provide JSON, but enables communication to the lock. If 5 log entries are requested, it should return 5 objects with log information. In your case it returns 22, which are then serialized to JSON. Since the library worked fine until now, I suspect  firmware bug in the 4.0 locks.

micronax, 2024-01-26:
But since the JSON itself is prematurely terminated and therefore no valid JSON, it could also be a memory / buffer problem in your JSON serialization, can't it?

Am I thinking to primitive or might it be just possible to discard any entries you receive from that lib after the fifth one and then just serialize those 5 ones?

technyon, 2024-01-27:
Well, on a microcontroller resources are limited. You can't just happily create strings like on a x86 architecture, since memory is limited. For serialization a static buffer is allocated, and the number of log entries was chosen accordingly so they fit into this buffer. NUKI Hub relies on a the library to only send it 5 entries, which in turn relies on the lock to send only 5 entries. The lock has a bug, sends more entries, the serializer stops when it reaches the buffer limit and your string is truncated.

Long story short, one should check external input, and yes maybe adding a check to only serialize the first five entries is a idea. Check the binary below.

[nuki_hub-8.30-pre-3.zip](https://github.com/technyon/nuki_hub/files/14070749/nuki_hub-8.30-pre-3.zip)

micronax, 2024-01-27:
Yes, I must admit that I don't have much experience with embedded and microcontroller coding.

But whatever you did now, worked. At least the JSON in that `log` field is not truncated anymore. Thank you 🙏 

But: It also doesn't show the most recent logs. I think the logs are being red in reverse order.

Currently the JSON contains the indices 2-6 which represents the oldest entries 2-6 shown in the app. So the first one is skipped, then the 5 next entries are shown. But since there the log recorded like ~25 more log entries.

technyon, 2024-01-27:
Ultimately NUKI has to fix this. The order to send this is also specified when sending the request to the lock, and the lock should return entries in the correct order.

iranl, 2024-05-19:
Fixed in #368

closed 2024-05-21 by technyon