technyon/nuki_hub #271: Stops reacting after some hours

Mikescotland, 2024-01-16:
Does anyone have similar problem?
After NH restart works fine, then randomly after a few hours stops communicating with Nuki lock, restart helps.
Nuki lock 3, latest firmware, NH also latest.

Mikescotland, 2024-01-17:
NUKI Hub version: 8.29
run: true
deviceId: 781432325
deviceIdOp: 3743102218
mqttbroker: 192.168.1.10
mqttport: 1883
mqttuser: ***
mqttpass: ***
mqttlog: false
lockena: true
mqttpath: nuki
openerena: false
mqttoppath: 
maxkpad: 
opmaxkpad: 
mqttca: 
mqttcrt: 
mqttkey: 
hassdiscovery: homeassistant
dhcpena: false
ipaddr: 192.168.1.33
ipsub: 255.255.255.0
ipgtw: 192.168.1.1
dnssrv: 192.168.1.1
nwhw: 1
rssipb: 60
hostname: nuki-hub
nettmout: -1
restdisc: true
rstbcn: -1
lockStInterval: 1800
configInterval: 3600
batInterval: 1800
kpInterval: 1800
kpEnabled: false
accLvl: 0
regAsApp: false
nrRetry: 5
rtryDelay: 100
crdusr: 
crdpass: 
pubauth: false
pubdbg: false
prdtimeout: 60
hasmac: false
macb0: 
macb1: 
macb2: 
MQTT connected: Yes
Lock firmware version: 3.7.7
Lock hardware version: 6.11
Lock paired: Yes
Lock PIN set: Yes
Lock has door sensor: No
Lock has keypad: No
Network device: Built-in Wifi
Uptime: 659 minutes
Heap: 75792
Stack watermarks: nw: 6008, nuki: 552, pd: 224
Restart reason FW: NotApplicable
Restart reason ESP: ESP_RST_PANIC: Software reset due to exception/panic.

technyon, 2024-01-17:
Try to enable "Restart if bluetooth beacons not received", you can set a value like 60 seconds.

Mikescotland, 2024-01-18:
Enabled. Will watch

Mikescotland, 2024-01-18:
And normally what is the interval lock usually sends the beacon?

technyon, 2024-01-18:
It depends on the energy saving settings you can set in the app ... a lot less than 60 seconds :). Maybe something like once per second.

Mikescotland, 2024-01-18:
Set it to fast in the app.
I'll monitor. Thanks

technyon, 2024-01-18:
No need to set it to fast. This is setting is more about what happens on ESP side. Unfortunately the bluetooth stack tends to die sometimes, this is unfortunately out of my control. At this point the ESP can't receive beacons anymore, and enabling this setting will cause it to reboot to revive the bluetooth stack.

Mikescotland, 2024-01-18:
So medium (or automatic) would be fine?
How does it affect the reaction time for let's say unlock?
Maybe it's a placebo effect but I think I noticed the status of the lock
gets updated quicker?

technyon, 2024-01-19:
@KingSausage Probably yes, it sounds similar.

There no effect in terms of how fast an action is executed. Beacons are only for communicating from lock to ESP. If you choose the fast setting, the ESP may catch a new state (e. g. you unlocked via the smartphone app) a little faster.

Mikescotland, 2024-01-20:
@technyon what would you recommend for esp32 with LAN for nuki hub? I'd like to remove wifi from this equation.

technyon, 2024-01-21:
The M5Stack Atom PEO or one of the Olimex boards. The WT32-ETH01 and M5Stack PoESP32 Unit work just fine too, but are more difficult to flash since you have to connect a USB-Serial converter to the right PINs, and even then it's tricky to get it to flash.

technyon, 2024-02-04:
@Mikescotland Were you able to try one the suggested boards?

Mikescotland, 2024-02-04:
Not yet, unfortunately
However suddenly the lock for over a week started to behave flawless! 😀

technyon, 2024-03-02:
@Mikescotland I assume it works fine for now, so I'm closing here. If not please follow up in issue #236

closed 2024-03-02 by technyon